cms-gem-daq-project/gem-plotting-tools #8: Feature Request: Summary Plot from anaUltraLatency.py

bdorney, 2017-07-24:
Right now anaUltraLatency.py plots the data as 24 different plots instead of a single summary plot.

This makes it hard for the DOC or analyst to upload the data to the E-Log for an "at-a-glance" view.

Request this script makes a summary plot.

mexanick, 2017-07-24:
I have requested modification provided by Alice Magnani. I'll prepare a PR

bdorney, 2017-08-07:
@mexanick PR #7 has been merged in, I think it would be best if I close this issue and we make a new issue/feature request to include the feedback/modification you mentioned from Alice.

Please advise.

bdorney, 2017-08-15:
Closing issue as no longer appears relevant and functionality implemented in merged PR.

closed 2017-08-15 by bdorney