cms-gem-daq-project/gem-plotting-tools #20: anaUltraScurve doesn't take the -o option into account

lmoureaux, 2017-08-23:
`anaUltraScurve.py` has a `-o` option to specify the output filename, but will always use the default.

## Brief summary of issue

The `-o` option is defined in `anaoptions.py` and not used by anaUltraScurve. It can be useful e.g. when willing to compare S-curve analysis at different `z-trim` or when debugging changes to `anaUltraScurve` itself.

### Types of issue

- [x] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

Passing a value to the `-o` option should change the output filename, or the `-o` option shouldn't appear in the help.

## Current Behavior

### Steps to Reproduce (for bugs)

1. Take an S-curve
2. Analyze the data using `-o test`
3. The analysis output is written to directory `SCurveData/`

## Possible Solution (for bugs)

When specified, let `-o` override the output directory.

## Context (for feature requests)

Was trying to find out if my code changes caused some weird behaviour in http://cmsonline.cern.ch/cms-elog/1005623 by comparing the `anaUltraScurve` output with and without my changes.

## Your Environment

* Version used: master and custom
* Shell used: bash

bdorney, 2017-08-23:
Not sure I see how this is possible in `anaUltraScurve.py` given that `options.outfilename` is specifically used:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/a81799b1b77ff69bfd145498565c7958c2752be1/anaUltraScurve.py#L30

And then applied at:
https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/a81799b1b77ff69bfd145498565c7958c2752be1/anaUltraScurve.py#L41

When using the single character option I believe you should not include a space, e.g.:

```
anaUltraScurve.py ... -otest
```

Whereas if you use the other method you would do: `--outfilename=test`.

Moreover what do you mean by:

> when willing to compare S-curve analysis at different z-trim

The s-curves are taken at a **specific** z-trim value.  So I understand that attempting to analyze the same scan data with at different z-trim values produces nonsense.

Finally the default filename behavior of output data files should not be changed based on the design of:
- `ana_scans.py`
- `vfatqc-python-scripts/confAllChambers.py`

Otherwise this will leave the user unable to use these tools to analyze data or configure detectors (both of which are specified by the dictionaries in `gem-plotting-tools/mapping/chamberInfo.py`.

lmoureaux, 2017-08-23:
> Not sure I see how this is possible in anaUltraScurve.py given that options.outfilename is specifically used
> Finally the default filename behavior of output data files should not be changed

Good points. Closing.

> The s-curves are taken at a specific z-trim value. So I understand that attempting to analyze the same scan data with at different z-trim values produces nonsense.

Meant `z-score`. Again. Modified first post.

closed 2017-08-23 by lmoureaux