cms-gem-daq-project/gem-plotting-tools #22: Feature Request: Link Functionality to mapping/chamberInfo.py

bdorney, 2017-08-23:
## Brief summary of issue

We have two test stands and P5.  Right now `mapping/chamberInfo.py` has commented lines for the two test stands and uncommented lines for P5.  When using a different test stand the user has to change these lines to match.

Suggest instead to implement the same idea as the connections file that we have in `cmsgemos`, like:

```
% ls -lh mapping/chamberInfo*
mapping/chamberInfo_904.py
mapping/chamberInfo_p5.py
mapping/chamberInfo.py -> chamberInfo_p5.py
mapping/chamberInfo.pyc
mapping/chamberInfo_qc8.py
```

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

We should have one symlink for `mapping/chamberInfo.py` which defaults to be the p5 file `chamberInfo_p5.py` and then two additional files:

1. GEM-CSC Integration stand: `chamberInfo_904.py`
2. Cosmic Stand: `chamberInfo_qc8.py`

Then if you are using one of these two stands you just have to change where the symlink points to rather than continuously editing the `mapping/chamberInfo.py` file.

## Current Behavior

Right now you have to continually edit the `mapping/chamberInfo.py` file when not at P5 or when switching test stands.

## Context (for feature requests)

I have to always edit the `mapping/chamberInfo.py` file and it is annoying and leads to lost time/mistakes.

jsturdy, 2017-08-30:
The stuck check has given me time, as I'm still pondering whether I like this solution to this "issue".
* I understand that the current hack being implemented in `cmsgemos` seems like a nice reason to implement the same hack here, however, the RPM build process of `cmsgemos` will create a machine/setup profile, making sure that the appropriate files are installed, so something similar could be envisioned here
 * Though the simplest in that case would be migrating `mapping/chamberInfo.py` and friends (or anything that is "setup specific" to the `gempython` package and letting the `cmsgemos` RPM build process take care of the profiles, but this, as below, requires more careful work as multiple changes will be required
* A single file, with a key for each setup is more desirable (which is how I've been working with `cmsgemos` lately), but this definitely requires some rework on other code, so I can appreciate the hack solution implemented
 * The longer term solution is probably to have this information in the(some) DB, and I'd probably prefer to spend the time to get that done correctly (much of this info is/*should* already be in the DB at P5, so we'd just need the instance in CERN IT for the rest of the chambers)

bdorney, 2017-08-30:
I'm not really knowledgable about the RPM build process so I don't know if I could comment effectively on the strengths/weaknesses of the first two bullet points.

> A single file, with a key for each setup is more desirable (which is how I've been working with cmsgemos lately), but this definitely requires some rework on other code, so I can appreciate the hack solution implemented

I think this would work for situations that are static, e.g. P5 and 904.  However during full production and QC of the detectors we will be cycling chambers on and off the cosmic stand routinely.  These will not use a position based naming schema but follow the serial number of the superchamber, so I envision chamber_config looking something like:

```
0:"GE11-SC-0001-L1"
1:"GE11-SC-0001-L2"
3:"GE11-SC-0002-L2"
etc...
```

Moreover this will need to be constantly changed (either daily or weekly depending on the measurement program).

> The longer term solution is probably to have this information in the(some) DB, and I'd probably prefer to spend the time to get that done correctly (much of this info is/should already be in the DB at P5, so we'd just need the instance in CERN IT for the rest of the chambers)

I think this is probably the best long-term solution.  Since this will track the static nature of P5 and also the shifting link names on the cosmic stand in a more effective manner.

However, for the short-term I would propose that the solution presented here be adopted.

jsturdy, 2017-08-30:
For "static" (or mostly) setups, the short term solution can be adapted.
But your comment about the "constantly changed" is exactly the reason I don't want to go down this road.  For the QC8 setup, I don't want to get into the game of changing this file and pushing to github regularly, mostly because I want to move away from the setup dependent aspects of the package (and elsewhere as well).
* If a setup needs to modify this file regularly, then so be it, but it should stay local (until a DB solution can be adopted)

bdorney, 2017-08-30:
I'm not proposing to constantly push to github whenever the qc8 file changes; Think of this as a file that can be changed by the user of the stand on either the user account of daq machine or the expert's own area and then allow easy switching between other areas.

I agree that a longterm solution involving the DB is best.  However I think this solution in the short term meets the "quality of life" improvement goal.

bdorney, 2018-03-15:
Closing as stale.

closed 2018-03-15 by bdorney