cms-gem-daq-project/gem-plotting-tools #37: Bug Report: anaUltraScurve.py still has event count hard coded in one spot

bdorney, 2017-09-14:
## Brief summary of issue

In `anaUltraScurve.py` the value for `events/2` is still harded at line [151](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/164fcef4e5ddc1f930cf70a7d82f376bebff6390/anaUltraScurve.py#L151).  Here it is set as `500` but should be an input parameter from the input TTree.

This should be changed since if a user takes an scurve with non-default settings (nevt=1000) this part of the code may not work as intended.

Low priority since this will work on scans taken with default parameters.

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

Value for `events/2` at line 151 should not be hard coded but instead be an input parameter.

## Current Behavior

Value is hard coded at `events/2 = 500` at line 151.

## Possible Solution (for bugs)

Change Line 151 to take an input instead of a hard coded value.

## Your Environment

* Version used: 164fcef4e5ddc1f930cf70a7d82f376bebff6390
* Shell used: /bin/bash

cbravo135, 2017-09-19:
I do think this is the wrong way to interpret this bug. We should change the scurve object to be a TGraphAsymmErrors made by dividing 2 histograms so C-P confidence intervals are used. Then all fits will be done with an efficiency of 0 to 1 and have the proper error bars. This would also mean several scurves could be statistically summed together and the analysis would be robust. It would also allow the number of pulses per Vcal bin to be unequal and the fit could still work.

bdorney, 2017-09-19:
I think this is a good suggestion, but I think this is enough of a change where it should be it's own separate issue.  For now the proposed solution will serve as a "short term" fix.

closed 2017-09-19 by bdorney