cms-gem-daq-project/gem-plotting-tools #49: Generic plotting tool calibration

cbravo135, 2017-10-17:
Currently all of the plots generated by the generic plotting tool gives all results in DAC units.
We should give this tool access to calibration parameters so plots can be reported in physical
units, rather than digital units. These plots ultimately should be in physical units so it is easier
for the user to compare results from VFAT to VFAT.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ x] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior
Plots are in simple units kept in the data format, and we need a way of building these
top level plots in physical units.

## Possible Solution (for bugs)
It is possible we need to have this script query a database of calibration parameters.

## Context (for feature requests)
Producing plots which have physical units allows the user to make meaningful comparisons
within the system. As is, the plots made by this tool can themselves not even be compared
to each other in a sensible way.

bdorney, 2017-10-17:
Following this DB idea it may be a good idea to add a TBranch called `ChipID` which stores the hex chipID as an `int` to all our TTree's both at the "raw" and "ana" levels.  This would ensure we pair the right info to each VFAT.

cbravo135, 2017-10-17:
Agreed.

bdorney, 2017-10-17:
To ensure github issue linking:
https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/50
https://github.com/cms-gem-daq-project/vfatqc-python-scripts/issues/138

bdorney, 2017-12-07:
Closing due to https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/54.  If additional TBranches are requested to be stored in DAC units please reopen this issue.

closed 2017-12-07 by bdorney