cms-gem-daq-project/gem-plotting-tools #56: Issue: plotTimeSeries.py is not described in README.md

bdorney, 2017-11-15:
## Brief summary of issue

I didn't see any documentation on for [plotTimeSeries.py](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/macros/plotTimeSeries.py) in the `README.md` or on the DOC Twiki page.  Ideally this should go here in the README.md so we don't lose track of it.

### Types of issue

- [X] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

`README.md` includes some explanation about the command line arguments and an example of the input file and usage. 

## Current Behavior

No documentation in the `README.md`

## Context (for feature requests)

Would helps the DOC to generate weekly calibration plots.

mexanick, 2017-11-16:
addressed in https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/57

Please approve the PR and close the issue

bdorney, 2017-11-16:
Documentation was present, I just missed it; however @mexanick added a link in the `README.md` TOC so it is more quick reference.  Closing issue.

closed 2017-11-16 by bdorney