cms-gem-daq-project/gem-plotting-tools #61: "Bug" Report: Error Bars on anaUltra*.py Scripts

bdorney, 2018-01-17:
## Brief summary of issue

We've migrated all our DAQ (and I guess also DQM?) machines to cc7 which defaults to ROOT v6.X.  In this version of ROOT by default TH1::Sumw2() is automatically set, see the [TH1 Class Documentation](https://root.cern.ch/doc/master/classTH1.html#a8cd3b44217c8fec1d74046ed2ac153c2).  This causes error bars of some of our histograms to not be set correctly.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

For a counting experiment (almost all of our scan routines) the error bar should go as `sqrt(N)` where `N` is the bin content.

## Current Behavior

Due to the differences between ROOT 6.X and 5.34 the `TH1::Sumw2()` method is automatically called.

Our analysis routines usually use the [Int_t TH1::Fill(Double_t x,Double_t w )](https://root.cern.ch/doc/master/classTH1.html#a498de8e0804e75fc75e62dc14a3bb62d) method for storing values in the histogram where `x` and `w` come from the input `TTree` object.  However this is only being called once per bin so the error bar ends up equalling `w` e.g. the sum of weights squared is a sum over one number.

See for example:

- anaUltraLatency.py [Line 55](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6f831848c4a39298a8d700e45eee87f58bed7838/anaUltraLatency.py#L55)
- anaUltraScurve.py [Line 249](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6f831848c4a39298a8d700e45eee87f58bed7838/anaUltraScurve.py#L249) 
- anaUltraThreshold.py [Line 112](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6f831848c4a39298a8d700e45eee87f58bed7838/anaUltraThreshold.py#L112)
 so in all instances where we are filling 

### Steps to Reproduce (for bugs)

1. Run any anaUltra script on a machine where ROOT is v6.X or higher.
2. Look at the error bars on the produced plots

## Possible Solution (for bugs)

### Proposal 1
Call [TH1::SetDefaultSumw2(False)](https://root.cern.ch/doc/master/classTH1.html#a8cd3b44217c8fec1d74046ed2ac153c2
) at the start of each script.  Although this is a static ROOT function and I'm not sure how that will play in python.

If it works this is the simplest fix.

### Proposal 2
Use `event.indepVar` to determine the corresponding bin in the histogram, e.g.

```
for idx in range(1, Histo.GetNbinsX()+1):
    if (Histo.GetBinLowEdge(1) + idx*Histo.GetBinLowEdge(1) ) <= event.indepVar and indepVar < (Histo.GetBinLowEdge(1) + (idx+1)*Histo.GetBinLowEdge(1) ):
        Histo.SetBinContent(idx, event.depVar)
```

Disadvantage is that this is pretty slow...

### Proposal 3
Change all `TH1` objects to `TGraph` objects (or inherited classes, e.g. `TGraphErrors` or `TGraphAsymErrors`) and use the `TGraph::SetPoint(...)` method for plotting from the `TTree`.  

If you draw the `TGraph` object you could then use the `TGraph::GetHistogram()` method to return a TH1 if it's absolutely necessary.

## Context (for feature requests)

This is particularly a problem when we are fitting distributions since the error bars on the data points will be accounted in the fitting and having each point as `N +/- N` is probably not going to give good fit results.

## Your Environment

* Version used: I think any commit will have this problem
* Shell used: `zsh`

jsturdy, 2018-01-17:
Regarding proposal 1: Using `r.TH1.SetDefaultSumw2(False)` will work in `python`

bdorney, 2018-03-28:
There should be one spot in `fitScanData.py` that should also have the error bars updated:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/75ca966d5226ced8874902923e269bd465fe28b2/fitting/fitScanData.py#L77

I would suggest fixing this by finding the `chargeBin` as was done in `anaUltraScurve.py`.  To do so you'll need to get the list of bin edges in terms of charge, e.g. replace `Y` with `X`: 
https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/75ca966d5226ced8874902923e269bd465fe28b2/anaUltraScurve.py#L41-L44

Then `chargeBin` can be determined via::
https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/75ca966d5226ced8874902923e269bd465fe28b2/anaUltraScurve.py#L66