cms-gem-daq-project/gem-plotting-tools #62: Bug Report: ScanDataFitter Generates Null Pointer For Empty TH1 in ROOT v6.X

bdorney, 2018-01-17:
## Brief summary of issue

Attempting to fit an scurve with 0 entries causes a null pointer `TFitResultPtr` object to be returned even when supplying the fit option `S`.

Previously it looks like in ROOT v5.34 a `TH1` object with 0 entries was being fit by a `TF1` object using the `TH1::Fit(...)` method it would return a non-null pointer to a `TFitResultPtr` object when the fit option `S`.  However, now in ROOT v6.X this returns a null pointer which crashes the program.

Looks like again default behavior of ROOT has changed between v5.34 and v.6.X.

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

If an scurve is empty it should just be skipped.

## Current Behavior

At [Line 80](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6f831848c4a39298a8d700e45eee87f58bed7838/fitting/fitScanData.py#L80) of `fitScanData.py` the object `fitResult` will be a null-pointer in ROOT v6.X if the `TH1` object being fit is empty even though the fit option `S` is used.

This then generates an error condition.

### Steps to Reproduce (for bugs)

1. Using ROOT v6.X
2. Mask a channel or use a VFAT with a dead channel
3. Run `ultraScurve.py`
4. Try to analyze the output with the fitting option `-f`

## Possible Solution (for bugs)

Since this is generated from a projection of a 2D histogram the `TH1::GetEntries()` method is not reliable to determine if it's empty or not.

One should instead rely on the Integral (since it will always be greater than 0 for an S-curve).

Specifically I'm proposing to change [Lines 59-60](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6f831848c4a39298a8d700e45eee87f58bed7838/fitting/fitScanData.py#L59) to the following:

```
if self.isDead[vfat][ch]:
    continue # Don't try to fit dead channels
elif not (self.scanHistos[vfat][ch].Integral() > 0):
    continue
```

## Context (for feature requests)

Fitting s-curve runs with empty s-curves will cause a crash on machines using ROOT v6.X.

## Your Environment

* Version used: probably any
* Shell used: `zsh`

jsturdy, 2018-01-29:
Looking at this in the `slc6` version at p5, we get warnings (while trimming) like:
```
Error in <TFitResultPtr>: TFitResult is empty - use the fit option S
```
so I suspect an inconsistent set of fitting options.
* Making this uniform everywhere, i.e., adding the `S` option to fits, and then checking that the `TFitResultPtr` is valid would eliminate crashes (something that is probably lost when writing `python` code is that one should *still* always be checking when a pointer is returned before trying to do anything with that pointer)
* This check proposed can possibly still be valuable in not wasting `CPU` resources on fits that will be guaranteed to fail, so the two checks probably need to be done in concert

jsturdy, 2018-03-12:
Provide `wget`able test data for test scripts to run in `travis`/`gitlab-ci`

bdorney, 2018-03-13:
> Provide `wget`able test data for test scripts to run in `travis/gitlab-ci`

You would like this uploaded to the repository under a `.travis` directory?  If so which branch should the PR be made to?  Maybe your private branch?

Right now `ScanDataFitter` is only being used by `trimChamber.py` of `vfatqc-python-scripts` and `anaUltraScurve.py`.  If you are asking for test data for all analysis tools in this repo than we should open a separate issue for tracking purposes.

bdorney, 2018-03-14:
Regarding the statements above:

> Looking at this in the slc6 version at p5, we get warnings (while trimming) like:
> Error in <TFitResultPtr>: TFitResult is empty - use the fit option S
> so I suspect an inconsistent set of fitting options.

> Making this uniform everywhere, i.e., adding the S option to fits, 

The fit option already includes `S` by default as can be seen:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/13607095bcc0895f3a2e34c342644a10400d5fd9/fitting/fitScanData.py#L80

>and then checking that the TFitResultPtr is valid would eliminate crashes (something that is probably lost when writing python code is that one should still always be checking when a pointer is returned before trying to do anything with that pointer)

Will python be able to detector a null pointer here? Or because it's a wrapper of a ROOT object it won't be known whether or not it is null until it is accessed?  Specifically I thought I tried in the past:

```
if fitResult is None:
   break
```

But I don't recall this solving the issue...can try again at some point.

bdorney, 2018-03-15:
Closing, resolved by #70

closed 2018-03-15 by bdorney