cms-gem-daq-project/gem-plotting-tools #63: Bug Report: gemSCurveAnaToolkit.py generates crash with ROOT v6.X or on cc7?

bdorney, 2018-01-18:
## Brief summary of issue

`gemSCurveAnaToolkit.py` generates a crash when running on cc7 with ROOT v6.X.  Suspect it's a difference between ROOT 5.34 and 6.X, unclear why.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior

### On `gem904daq03`
Works
```
[dorney@gem904daq03]~/scratch0/CMS_GEM/CMS_GEM_DAQ/gem-plotting-tools% gemSCurveAnaToolkit.py -i listOfScanDates_Scurve.txt -c -s30 -v1 --anaType=scurveAna --drawLeg
TClass::TClass:0: RuntimeWarning: no dictionary for class TF1Parameters is available
TStreamerInfo::BuildOld:0: RuntimeWarning: Cannot convert TF1::fParErrors from type:vector<double> to type:Double_t*, skip element
TStreamerInfo::BuildOld:0: RuntimeWarning: Cannot convert TF1::fParMin from type:vector<double> to type:Double_t*, skip element
TStreamerInfo::BuildOld:0: RuntimeWarning: Cannot convert TF1::fParMax from type:vector<double> to type:Double_t*, skip element
TStreamerInfo::BuildOld:0: RuntimeWarning: Cannot convert TF1::fSave from type:vector<double> to type:Double_t*, skip element
TStreamerInfo::BuildOld:0: RuntimeWarning: Cannot convert TF1::fParams from type:TF1Parameters* to type:Double_t*, skip element
Info in <TCanvas::Print>: png file canv_SCurveFitOverlay_VFAT1_Chan30.png has been created
eog canv_SCurveFitOverlay_GE11-X-S-CERN-0003_2018.01.17.14.29_VFAT1_Chan30.png

Your plot is stored in a TFile, to open it execute:
root /afs/cern.ch/user/d/dorney/scratch0/CMS_GEM/CMS_GEM_DAQ/data/gemelog/gemSCurveAnaToolkit.root
```

ROOT is 5.34/36
python is Python 2.6.6

### On `gem904qc8daq`
```
[dorney@gem904qc8daq]~/scratch0/CMS_GEM/CMS_GEM_DAQ/gem-plotting-tools% gemSCurveAnaToolkit.py -i listOfScanDates_Scurve.txt -c -s30 -v1 --anaType=scurveAna --drawLeg

 *** Break *** segmentation violation

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================
#0  0x00007fe88412bdbc in waitpid () from /lib64/libc.so.6
#1  0x00007fe8840aecc2 in do_system () from /lib64/libc.so.6
#2  0x00007fe87c3041a2 in TUnixSystem::StackTrace() () from /usr/lib64/root/libCore.so.6.10
#3  0x00007fe87c306b3c in TUnixSystem::DispatchSignals(ESignals) () from /usr/lib64/root/libCore.so.6.10
#4  <signal handler called>
#5  0x00007fe86f47289f in TF1::GetParameters() const () from /usr/lib64/root/libHist.so.6.10
#6  0x00007fe86f46904b in TF1::DoCreateHistogram(double, double, bool) () from /usr/lib64/root/libHist.so.6.10
#7  0x00007fe86f46889d in TF1::Paint(char const*) () from /usr/lib64/root/libHist.so.6.10
#8  0x00007fe86fbed5ac in TPad::Paint(char const*) () from /usr/lib64/root/libGpad.so.6.10.08
#9  0x00007fe86fbeb84f in TPad::Print(char const*, char const*) () from /usr/lib64/root/libGpad.so.6.10.08
#10 0x00007fe86fbec86f in TPad::SaveAs(char const*, char const*) const () from /usr/lib64/root/libGpad.so.6.10.08
#11 0x00007fe87ce79bd7 in FastCall(long, void*, void*, void*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#12 0x00007fe87ce7ea0f in PyROOT::TVoidExecutor::Execute(long, void*, PyROOT::TCallContext*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#13 0x00007fe87cea239d in PyROOT::TMethodHolder::CallSafe(void*, long, PyROOT::TCallContext*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#14 0x00007fe87cea0d29 in PyROOT::TMethodHolder::Execute(void*, long, PyROOT::TCallContext*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#15 0x00007fe87ce9ff1e in PyROOT::TMethodHolder::Call(PyROOT::ObjectProxy*&, _object*, _object*, PyROOT::TCallContext*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#16 0x00007fe87ce845ee in PyROOT::(anonymous namespace)::mp_call(PyROOT::MethodProxy*, _object*, _object*) () from /usr/lib64/python2.7/site-packages/libPyROOT.so
#17 0x00007fe884da09a3 in PyObject_Call () from /lib64/libpython2.7.so.1.0
#18 0x00007fe884e350f6 in PyEval_EvalFrameEx () from /lib64/libpython2.7.so.1.0
#19 0x00007fe884e3befd in PyEval_EvalCodeEx () from /lib64/libpython2.7.so.1.0
#20 0x00007fe884e393fc in PyEval_EvalFrameEx () from /lib64/libpython2.7.so.1.0
#21 0x00007fe884e3befd in PyEval_EvalCodeEx () from /lib64/libpython2.7.so.1.0
#22 0x00007fe884e3c002 in PyEval_EvalCode () from /lib64/libpython2.7.so.1.0
#23 0x00007fe884e5543f in run_mod () from /lib64/libpython2.7.so.1.0
#24 0x00007fe884e565fe in PyRun_FileExFlags () from /lib64/libpython2.7.so.1.0
#25 0x00007fe884e57889 in PyRun_SimpleFileExFlags () from /lib64/libpython2.7.so.1.0
#26 0x00007fe884e68a3f in Py_Main () from /lib64/libpython2.7.so.1.0
[... 136 lines cut ...]
1. From new `cc7` machine
2.  Source gem-plotting-tools env
3. Execute `gemSCurveAnaToolkit.py -i listOfScanDates_Scurve.txt -c -s30 -v1 --anaType=scurveAna --drawLeg`

Contents of `listOfScanDates_Scurve.txt` is reported below:

```
ChamberName	scandate
#GE11-VI-L-CERN-0002	current
#GE11-X-S-CERN-0001	current
GE11-X-S-CERN-0003	2018.01.17.14.29
```

## Possible Solution (for bugs)

Unclear.

## Context (for feature requests)

Cannot investigate individual channel scurves and their fits. 

## Your Environment

* Version used: Probably all of them
* Shell used: `zsh`

jsturdy, 2018-03-12:
Provide `wget`able test data for test scripts to run in `travis`/`gitlab-ci`

bdorney, 2018-03-13:
> Provide `wget`able test data for test scripts to run in `travis/gitlab-ci`

You would like this uploaded to the repository under a .travis directory? If so which branch should the PR be made to? Maybe your private branch?

jsturdy, 2018-03-14:
No, I specifically don't want to put this test data in the repo, I just want a location that the CI can `wget` a file for testing from.
A possible alternative would be putting the test data in a dedicated docker container
The last ditch effort would be to put the data in the repo

mexanick, 2018-03-14:
Another option is to add test data as attached file to the git release. Just my 2cents

bdorney, 2018-03-14:
> No, I specifically don't want to put this test data in the repo, I just want a location that the CI can wget a file for testing from.

I don't understand then, I thought `wget` requires a url?  I could attach a test file to this issue page and then `wget` could use that?  Is this the request?

> A possible alternative would be putting the test data in a dedicated docker container

I don't know what a "docker container" is; is this something that is uploaded to travis?  Sorry it's my ignorance here.

> The last ditch effort would be to put the data in the repo

I think Misha's suggestion as having the files uploaded to a release is a good one, this then gives the user a test sample to play with if necessary.

bdorney, 2018-03-14:
It might be worth pointing out that the `gemSCurveAnaToolkit.py` is also expecting the file format of which is identical to how `$DATA_PATH` is organized, as described in the documentation in the `README.md` [here](https://github.com/cms-gem-daq-project/gem-plotting-tools/#gemtreedrawwrapperpy-input-file).

So in the example that is shown in the link above `gemSCurveAnaToolkit.py` is expecting a filestructure of the form:

```
$DATA_PATH/GE11-VI-L-CERN-0001/scurve/2017.08.11.16.30/ScurveData.root
$DATA_PATH/GE11-VI-L-CERN-0001/scurve/2017.08.14.20.54/ScurveData.root
$DATA_PATH/GE11-VI-L-CERN-0001/scurve/2017.08.30.15.03/ScurveData.root
$DATA_PATH/GE11-VI-L-CERN-0001/scurve/2017.08.30.21.39/ScurveData.root
$DATA_PATH/GE11-VI-L-CERN-0001/scurve/2017.08.31.08.28/ScurveData.root
``` 

What I would propose is that I provide the following:

- `listOfScanDates_Scurve.txt`
- tarball of `SCurveData.root` files in the appropriate filepath format

What may also be needed is a version of `$GEM_PLOTTING_PROJECT/mapping/chamberInfo.py` that has the following dictionaries compatible with the `listOfScanDates_Scurve.txt` specifically:

```
chamber_config = {
0:"GE11-VI-L-CERN-0001"
}

GEBtype = {
0:"long"
}
```

I can provide this to you and then you can choose how/where it should be hosted.

jsturdy, 2018-03-14:
> Another option is to add test data as attached file to the git release. Just my 2cents
Well, this doesn't fit in the "run standard tests model that is the point of this addition.
No reason they couldn't *also* be attached to a release, providing a set of test data for a given release, but that's not the use case I'm imagining for this.
Here we'd just have a script that runs some unit tests, it needs some data, this data can easily be provided by some URL (or set up inside the container in some standard setup), and every PR would get tested against this.
If the data needs to change, the test script gets updated...

> I can provide this to you and then you can choose how/where it should be hosted.
OK, think about how the unit tests will need to be run and what would need to be included in the tarball, then I can take it from there