cms-gem-daq-project/gem-plotting-tools #68: Structure of RPM package

jsturdy, 2018-03-10:
### Information
* The packages and modules will be packaged and end up in the system `PYTHONPATH`, e.g., (default `python` on `cc7):  `/usr/lib/python2.7/site-packages/`
  * I've been toying with adding a global namespace `gempython`, under which there would be the `gemplotting`, `vftaqc`, etc., modules, but this is still to be finalized into a proposal, and then discussed
  * This would necessarily involve a small redesign of the `import` statements, but I have for the most part decided on a workable solution for people who are doing private development and **not** using the production software
* Executable scripts, e.g., `amc_info_uhal.py` will **all** go into `/opt/cmsgemos/bin`, to make it simple to add this to the `PATH` variable and have the script ready

### Current RPM structure
```
rpm -q --filesbypkg -p rpm/gem_plotting_tools-0.0.1.dev172-centos7.python2.7.x86_64.rpm
gem_plotting_tools        /opt/gem-plotting-tools/bin/anaUltraLatency.py
gem_plotting_tools        /opt/gem-plotting-tools/bin/anaUltraScurve.py
gem_plotting_tools        /opt/gem-plotting-tools/bin/anaUltraThreshold.py
gem_plotting_tools        /opt/gem-plotting-tools/bin/anaXDAQLatency.py
gem_plotting_tools        /opt/gem-plotting-tools/bin/ana_scans.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/PKG-INFO
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/SOURCES.txt
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/dependency_links.txt
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/not-zip-safe
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/pbr.json
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools-0.0.1.dev172-py2.7.egg-info/top_level.txt
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/__init__.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/__init__.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/__init__.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/__init__.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/__init__.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/__init__.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/fitScanData.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/fitScanData.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/fitting/fitScanData.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/__init__.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/__init__.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/__init__.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemPlotter.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemPlotter.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemPlotter.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemSCurveAnaToolkit.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemSCurveAnaToolkit.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemSCurveAnaToolkit.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemTreeDrawWrapper.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemTreeDrawWrapper.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/gemTreeDrawWrapper.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/overlay_fit.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/overlay_fit.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/overlay_fit.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_eff.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_eff.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_eff.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_noise_vs_trim.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_noise_vs_trim.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_noise_vs_trim.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_scurves_by_thresh.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_scurves_by_thresh.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_scurves_by_thresh.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_and_channel_Scurve.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_and_channel_Scurve.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_and_channel_Scurve.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_summary.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_summary.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plot_vfat_summary.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plotoptions.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plotoptions.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/plotoptions.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/scurvePlottingUtitilities.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/scurvePlottingUtitilities.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/scurvePlottingUtitilities.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/summary_plots.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/summary_plots.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/macros/summary_plots.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/PanChannelMaps.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/PanChannelMaps.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/PanChannelMaps.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/__init__.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/__init__.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/__init__.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/buildMapFiles.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/buildMapFiles.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/buildMapFiles.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/chamberInfo.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/chamberInfo.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/chamberInfo.pyo
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/channelMaps.py
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/channelMaps.pyc
gem_plotting_tools        /usr/lib/python2.7/site-packages/gem_plotting_tools/mapping/channelMaps.pyo
```

bdorney, 2018-03-12:
Before we finalize the RPM for `cc7` I think it may be a useful idea to fix all ROOT 5.34 vs. 6.X.Y issues; I know benjamin was working on solving at least one.

The issues I'm referring too are:

https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/62
https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/63

jsturdy, 2018-03-12:
What is the usage of these:
```
ll gem-plotting-tools/macros
total 104
-rwxr-xr-x. 1 sturdy zh  1655 Mar  6 15:54 gemPlotterAllChannels.sh
-rwxr-xr-x. 1 sturdy zh 22749 Mar  6 15:55 gemPlotter.py
-rwxr-xr-x. 1 sturdy zh  6964 Mar  6 15:57 gemSCurveAnaToolkit.py
-rwxr-xr-x. 1 sturdy zh 16859 Mar  6 15:57 gemTreeDrawWrapper.py
-rw-r--r--. 1 sturdy zh     0 Mar  6 15:46 __init__.py
-rw-r--r--. 1 sturdy zh  1156 Mar  7 15:45 overlay_fit.py
-rwxr-xr-x. 1 sturdy zh  8028 Mar  6 15:57 plot_eff.py
-rwxr-xr-x. 1 sturdy zh   390 Mar  6 15:57 plot_noise_vs_trim.py
-rw-r--r--. 1 sturdy zh   651 Mar  6 15:46 plotoptions.py
-rw-r--r--. 1 sturdy zh  1092 Mar  6 15:57 plot_scurves_by_thresh.py
-rwxr-xr-x. 1 sturdy zh   376 Mar  6 15:57 plot_vfat_and_channel_Scurve.py
-rwxr-xr-x. 1 sturdy zh   252 Mar  6 15:57 plot_vfat_summary.py
-rw-r--r--. 1 sturdy zh  5211 Mar  6 15:57 scurvePlottingUtitilities.py
-rwxr-xr-x. 1 sturdy zh  4230 Mar  6 15:57 summary_plots.py
```

By which I mean
* Are they scripts that must be in the `PATH`? 
* Are they `import`ed in other tools?
* Are they a mishmash of these?

jsturdy, 2018-03-12:
Replying to @bdorney's comment, those bugs are not connected to this development.
Adding this step will allow for an `slc6`+`root-5.34` RPM to be built (which will be functional and is a good thing) as well as a currently broken `cc7`+default `root` RPM (which doesn't have to be installed anywhere)

bdorney, 2018-03-13:
Same question here as in `vfatqc-python-scripts`:

In the RPM structure above how are you handling: `$GEM_PLOTTING_PROJECT/mapping/chamberInfo.py`?

Here `ana_scans.py` requires the dictionaries that are defined and filled in this file. Additionally, it is expected that the user is able to change the contents of those dictionaries. While this will most likely not occur often for P5 in the present setup of the software it will most certainly be changing for QC8.

bdorney, 2018-03-13:
> What is the usage of these:
> ...
> ...
> By which I mean
> Are they scripts that must be in the PATH?

The following should be in the `$PATH`:

- `summary_plots.py`,
- `plot_vfat_summary.py`,
- `plot_vfat_and_channel_Scurve.py`,
- `plot_scurves_by_thresh.py` (might need to add executable permissions),
- `plot_noise_vs_trim.py`,
- `gemTreeDrawWrapper.py`,
- `gemPlotter.py`,
- `gemPlotterAllChannels.sh`,
- `gemSCurveAnaToolkit.py`, and
- `plot_eff.py`.

> Are they imported in other tools?

The following files have pieces that are imported by other tools:
- `plotoptions.py`, 
- `scurvePlottingUtitilities.py`, and
- `gemTreeDrawWrapper.py`.

> Are they a mishmash of these?

Yes as shown above.

jsturdy, 2018-03-13:
Let me clarify a bit:
* Which of these will a user call directly?
  * These are the ones that I would like to wind up in `/opt/cmsgemos/bin`
* Which of these are called indirectly, i.e., via another script?
  * These ones I would possibly prefer to end up in `/usr/lib/python<ver>/site-packages/gempython/<package>/{macros or bin}`, which could be appended (or prepended) to the `PATH`
  * Mostly, I don't want to clutter up `/opt/cmsgemos/bin`

Probably a request for a future improvement, but to properly package this as a python package, the separation between modules and scripts/executables should be exact, i.e., nothing that gets `import`ed (from) should stay in a script which is also executed.

See response in vfatqc-python-tools#160 for more info on handling the `import`s

bdorney, 2018-03-14:
> Which of these will a user call directly?

The following files:

- `summary_plots.py,
- `plot_vfat_summary.py`,
- `plot_vfat_and_channel_Scurve.py`,
- `plot_scurves_by_thresh.py` (need to add executable permissions, as a mistake it lacks them),
- `plot_noise_vs_trim.py`,
- `gemTreeDrawWrapper.py`,
- `gemPlotter.py`,
- `gemPlotterAllChannels.sh`,
- `gemSCurveAnaToolkit.py`, and
- `plot_eff.py`.

> Which of these are called indirectly, i.e., via another script?

- `plotoptions.py`,
- `scurvePlottingUtitilities.py`, and
- The function `getStringNoSpecials(...)` is imported from the file `gemTreeDrawWrapper.py`.

> Probably a request for a future improvement, but to properly package this as a python package, the separation between modules and scripts/executables should be exact, i.e., nothing that gets imported (from) should stay in a script which is also executed.

Based on this what I understand as your request is that the executable scripts only include a:

```
if __name__ == '__main__':
    # Code
```

And do not include additional function definitions.  For example in `gemTreeDrawWrapper.py` the follwoing functions are defined:

- `getStringNoSpecials(...)`, and
- `getPlotFromTree(...)`

You would prefer these functions be in a utilities or other file and imported here?  

Do I understand the request correctly:  Executables include only a `main` and no additional definitions/variables/functions/etc...

bdorney, 2018-03-14:
> See response in cms-gem-daq-project/vfatqc-python-scripts#160 for more info on handling the imports

Okay we'll move that discussion to that issue in `vfatqc-python-scripts`.

jsturdy, 2018-03-14:
> Do I understand the request correctly: Executables include only a main and no additional definitions/variables/functions/etc...

As a bare minimum, no additional function definitions that are used elsewhere, as it complicates the package organization.
If there's this nested mishmash, and only tools in the same directory cross-pollinate, then having a scripts/macros directory for this type of tools makes sense.  It has to then be added to the `PATH` but can be figured out in the packaging step

jsturdy, 2018-03-14:
> e.g. the other scripts like anaUltra*.py should also be placed in macros.

Hmm, I guess in this case the distinction I'm trying to raise is quite muddled...
How I initially imagined it, the "tools" themselves will go in `/opt/cmsgemos/bin` (here I view `ana<blah>.py` as the tools).
The `gemplotting` "scripts" (current random assortment of drawing/display scripts) will live in `/usr/lib/python<version>/site-packages/gempython/gemplotting/{macros or bin}` which would then be probably added to the `PATH`

Effectively what I want is to separate the "library" from the "things that use the library"

So probably in the short term, I'll put a select set of tools in `/opt/cmsgemos/bin` and leave the rest inside the package itself

bdorney, 2018-08-14:
I think it's safe to close this issue.  Reopen if still relevant.

closed 2018-08-14 by bdorney