cms-gem-daq-project/gem-plotting-tools #79: Feature Request: ana_scans.py uses **kwargs

bdorney, 2018-04-11:
## Brief summary of issue

The [ana_scans.py](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/ana_scans.py) tool has exceptionally bloated arguments and could benefit from use of **kwargs.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

Enhanced readability could be achieved by implementing `**kwargs`.

## Current Behavior

Lot of arguments in the command leave room for mistakes, especially if trying to supply them in order rather than by keyword.

## Context (for feature requests)

Maintenance.

## Your Environment

* Version used: `develop`
* Shell used: any

jsturdy, 2018-04-11:
While I mostly agree with the underlying point, currently, *only* `anaType, cName, cType, scandate` are required in a specific order, all others may be specified by keyword.
This bloat precipitated from the way arguments are passed to the thread pool, so as long as behaviour there is not impacted, this could be a trivial change
(Similarly, but admittedly less clean, in cms-gem-daq-project/vfatqc-python-scripts#162)

bdorney, 2019-05-20:
superceded by #202

closed 2019-05-20 by bdorney