cms-gem-daq-project/gem-plotting-tools #83: Feature Request: Algorithmic Identification of Burned/Disconnected VFAT2 Inputs

bdorney, 2018-04-16:
## Brief summary of issue

Based on [slide 22](https://indico.cern.ch/event/721622/contributions/2968019/attachments/1631961/2602748/BDorney_GEMDAQMtg_20180412_BurnedVFATInputs.pdf) `vfat2` channels that have been burned during operation have a characteristic s-curve width.  This s-curve width could be used to mark the channel as having a "burned input" and thus could be factored into our "% active channels" list when reporting to CMS.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

The `maskReason` class of `anaInfo.py` is updated to have a new field `DisconnectedInput` (it could also be called `BurnedInput` but we would need to know a priori if the channel was always like this or was assigned this classifier after some time).  This could be added:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6445dcb44ad50ef49fb2c48eed9e49171ce72861/anaInfo.py#L41-L54

Then in `anaUltraScurve.py` the "create reason array" block could be updated to search for channels with this behavior:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/6445dcb44ad50ef49fb2c48eed9e49171ce72861/anaUltraScurve.py#L361-L376

Here the solution would be something like:

```
reason[4.14E-02 < scanFitResults[1][vfat] and scanFitResults[1][vfat] < 1.09E-01] = MaskReason.DisconnectInput
```

Here the `4.14e-2` comes from the lower bound of the high purity burned sample and `1.09e-1` comes from the upper bound of the literature value.  Numbers taken from [slide 22](https://indico.cern.ch/event/721622/contributions/2968019/attachments/1631961/2602748/BDorney_GEMDAQMtg_20180412_BurnedVFATInputs.pdf). However these cut values could be influenced based on the result of #82.

## Current Behavior

Presently channels which have become burned, or were always disconnected, are erroneously included in our "live channel fraction" when reporting to CMS.

## Context (for feature requests)

Would like to correctly report our "live channel fraction" and monitor the progression of "burned inputs" over time in P5.

## Your Environment

* Version used: `develop`
* Shell used: any

bdorney, 2018-04-20:
Actually it seems that the `maskReason` of  `DeadChannel = 0x04` is never assigned.

We could start enforcing this reason for those channels that have burned inputs using the cut range described above, e.g.:

```
reason[4.14E-02 < scanFitResults[1][vfat] and scanFitResults[1][vfat] < 1.09E-01] = MaskReason.DisconnectInput
```

jsturdy, 2018-04-22:
OK, in this case, I would not make a distinction in the algorithm then.
If someone wants a breakdown of "Dead from the beginning" vs "Died during operation due to some issue", then a different tool/set of categories can be created

bdorney, 2018-04-22:
> OK, in this case, I would not make a distinction in the algorithm then.

Indeed, in the implementation here at https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/87 no distinction is made.

bdorney, 2018-04-24:
Addressed by #87

closed 2018-04-24 by bdorney