cms-gem-daq-project/gem-plotting-tools #94: Docs migration

jsturdy, 2018-05-15:
## Brief summary of issue
Current README needs to be refactored into a proper `docs` style
README should have info about getting set up, links to contributing guidelines, links to full documentation

### Types of issue
- [x] Cleanup

## Possible Solution (for bugs)
* Create `docs` folder
* Migrate documentation portion of README into this, preferably using rst for compatibility with `sphinx`/`readthedocs` style documentation

mexanick, 2018-05-18:
Yes, let's use sphinx as a baseline. I'll look into sphinx setup for the reg_utils repo and will provide more detailed feedback and maybe a sort of template here as well. Basically, sphinx is using the ReStructured Text as doc format which is reasonably close to Markdown used in the existing readme, so it would not be a hard task to get the proper documentation with it. We will need a skeleton first and then distribute the existing content around it. Given the completeness of this project documentation in a README, I do not expect a need to write a lot of new stuff. 
After this is done and RPM installation procedure is polished, the README should shrink to a short instruction on getting the code installed and link to the full documentation as @jsturdy suggested.

bdorney, 2018-06-28:
Was this resolved by #115 and #116?  Or is something more needed?

lmoureaux, 2018-07-09:
> Was this resolved by #115 and #116? Or is something more needed?

A significant part of the README and many docstrings still have to be ported to `rst`

bdorney, 2018-07-09:
Okay, thanks for update.  We keep this issue open for now.

PR #136 (2018-08-14): Docs migration: finish migration of the README

## Description

This PR migrates the remaining parts of the README to the Sphinx system.

It also changes the structure of the documentation to keep the table of contents readable.

This should be enough to close #94, even though:

* Part of the documentation can only be found on the DOC TWiki
* Not all comments in python modules have been converted into docstrings yet
* Sphinx is still unhappy about some constructs in the preexisting docstrings

### Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non [... cut]
closed 2018-08-14 by bdorney