cms-gem-daq-project/gem-plotting-tools #117: Bug Report: project does not build after Sphinx Inclusion

bdorney, 2018-06-21:
## Brief summary of issue

In commit 58e957b732116ab7078f8a913835fa9f7a13cee1 we added to the `Makefile`:

```
include $(BUILD_HOME)/$(Project)/config/mfSphinx.mk
```

This causes the project to not build.

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

Project should build successfully.

## Current Behavior

```
[dorney@gem904qc8daq]~/scratch0/CMS_GEM/CMS_GEM_DAQ/gem-plotting-tools% make clean
PythonModules=["gempython.gemplotting", "gempython.gemplotting.utils", "gempython.gemplotting.fitting", "gempython.gemplotting.macros", "gempython.gemplotting.mapping" ]
Using BUILD_HOME=/afs/cern.ch/user/d/dorney/scratch0/CMS_GEM/CMS_GEM_DAQ
OS Detected: centos7
Makefile:35: /afs/cern.ch/user/d/dorney/scratch0/CMS_GEM/CMS_GEM_DAQ/gem-plotting-tools/config/mfSphinx.mk: No such file or directory
make: *** No rule to make target `/afs/cern.ch/user/d/dorney/scratch0/CMS_GEM/CMS_GEM_DAQ/gem-plotting-tools/config/mfSphinx.mk'.  Stop.
```

### Steps to Reproduce (for bugs)

1. `git fetch -p gemdaq`
2. `git checkout develop`
3. `git pull gemdaq develop`
4. `make clean`

## Possible Solution (for bugs)

Unclear

## Context (for feature requests)

Prevents development

## Your Environment

* Version used: 50a2a91
* Shell used: `/bin/zsh`

jsturdy, 2018-06-21:
did you do `git submodule update`?

bdorney, 2018-06-21:
> did you do git submodule update?

Nope...and that solved the problem.

closed 2018-06-21 by bdorney

jsturdy, 2018-06-21:
FYI: from `git`version `2.13.1` (latest available on our `EPEL` repos is `2.12.X`...) there is the `--recurse-submodules` [option](https://www.mail-archive.com/git@vger.kernel.org/msg120829.html) which can also be set as a default [option](https://stackoverflow.com/a/44529991) (if desired)

I'll investigate a solution to get a later version on our DAQ machines