cms-gem-daq-project/gem-plotting-tools #137: Public documentation: ReadTheDocs (or other) integration

lmoureaux, 2018-07-17:
## Brief summary of issue

The documentation for the latest version should be built without human intervention.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

The public documentation is automatically rebuilt every time `develop` is updated or a new version is tagged.

## Current Behavior

The documentation has to be built manually, then copied to some public location such as http://homepage.iihe.ac.be/~lmoureau/GEM/sphinx/html/index.html.

## Possible Solution

Integrate the repository with an automatic system such as ReadTheDocs. Some changes are needed because AFAIK we can't install PyROOT on the doc builder machine.

## Context

The documentation has been migrated to Sphinx, but we can't really use it as long as it's not available on a public page. Updating it by hand is not a sustainable mode of operation.

jsturdy, 2018-07-17:
We will be using EOS served websites served with a CERN service account that is used for various GEM DAQ related tasks.
The easiest will probably be to do this via `gitlab-ci` as we'll have closer access to CERN resources there (this is also where the `yum` repository will be hosted)

The overall structure is not yet set up and is not quite finalized, but once it is, it will be integrated into the CI for the various packages

lpetre-ulb, 2020-05-18:
Documentation is pushed to the GEM DAQ test website and is about to be pushed to main GEM DAQ website.

closed 2020-05-18 by lpetre-ulb