cms-gem-daq-project/gem-plotting-tools #142: Relicensing as MIT (was GPLv3)

jsturdy, 2018-07-20:
### Brief summary of issue
Unfortunately, not enough thought (on my part) was used initially in selecting a license for several packages and it has resulted in different licenses being selected for packages that have no real reason to be differently licensed.  In the case of this package, contradictory information was packaged, an issue which was raised by @lmoureaux in #139 (thanks!)

As such, I am proposing changing the license of [`gempython_gemplotting`](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/LICENSE) (this repository) to an MIT license, to match the licensing done in [`gempython_vfatqc`](https://github.com/cms-gem-daq-project/vfatqc-python-scripts/blob/develop/LICENSE).

A further/secondary discussion (not in this thread) should be made regarding `xhal`, `reg_utils`, `cmsgmos_gempython`, which each state the MIT license in the python packaging metadata, but do not explicitly have a licence attached (no implicit expectation of a specific license from contributors)

### Types of issue
- [x] Permission request (request for confirmation by contributors)

### Contributors (as stated by github)
If the following people would write an explicit acceptance of the proposed license change, or provide explanation why they feel that keeping the GPLv3 is in their or the group's best interests, it would be much appreciated.  Additionally, if you know that there are other contributors whose agreement should be solicited, please advise

[x] @jsturdy
[ ] @cbravo135
[ ] @lmoureaux
[ ] @bdorney
[ ] @mexanick

bdorney, 2018-07-20:
Could we have a brief summary of what the change entails? E.g. what’s the difference?

jsturdy, 2018-07-20:
The links to the two licenses are provided in the second paragraph, from there one can read the full text, as well as the github page on choosing a license

lmoureaux, 2018-07-20:
Choice of license:

1. Why do you think the MIT is more appropriate than the GPL? How did you chose it over, say, LGPL or a BSD-style license?

Legal stuff (though IANAL):

2. Did you make sure that no GPL-licensed code taken from the Internet was included in the tree?
3. Did you check the list of contributors before tools were moved from `vfatqc-python-scripts` (e.g. PR #4)? It had no license at the time, so in principle we need explicit consent (even for MIT because it was added later).
4. The copyright for my contributions starting 1st of October, 2017 is assigned to [ULB](http://ulb.ac.be) as my employer. I don't know who should be contacted for copyright matters; probably the legal service.
5. The situation for my contributions before that date is unclear, since:
    1. I was a student at the time.
    2. Strictly speaking, my work at CERN was not part of the master programme at ULB.
    3. I couldn't have come to CERN without being a master student.