cms-gem-daq-project/gem-plotting-tools #154: Feature request: Sensible default for --calFile (or make sure it's provided when --VFAT3 is used)

lmoureaux, 2018-10-01:
## Brief summary of issue

When analyzing VFAT3 S-curves with `--isVFAT3`, the `--calFile` option is required. If it's not provided, the VFAT2 settings are used, which results in unusable output.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

`anaUltraScurve` checks its inputs for consistency and provide sensible defaults for optional arguments.

## Current Behavior

`anaUltraScurve` doesn't check that `--calFile` is present when `--isVFAT3` is used.

## Possible Solutions

1. Check that `--calFile` is present when `--isVFAT3` is used, or
2. Provide sensible defaults for the VFAT3 calibration

## Context (for feature requests)

See [this discussion](https://mattermost.web.cern.ch/cms-gem-ops/pl/r4fce9qo5in4jcs9pqnow4w3fe) on Mattermost

## Your Environment

* Version used: as installed at ULB setup on 2018/1/10
* Shell used: bash

bdorney, 2018-10-01:
As discussed on `Mattermost`, the intended use case with `vfat3` was that the `--calFile` would provide the exact calibration for each chip (or at least values that would approximate this).  Raw DAC units view is not supported.  If this is interested please see other plotting tools:

- [gemPlotter.py](https://github.com/cms-gem-daq-project/gem-plotting-tools#gemplotterpy), or
- [gemSCurveAnaToolkit.py](https://github.com/cms-gem-daq-project/gem-plotting-tools#gemscurveanatoolkitpy), or
- [gemTreeDrawWrapper.py](https://github.com/cms-gem-daq-project/gem-plotting-tools#gemtreedrawwrapperpy).

These will allow a "DAC unit" view.

bdorney, 2018-10-01:
> Version used: as installed at ULB setup on 2018/1/10

```bash
% yum info \*vfatqc\* | grep "Version"
Version     : 2.1.0
% yum info \*vfatqc\* | grep "Release"
Release     : centos7.python2.7
```

bdorney, 2018-10-14:
After further thought on this matter, the `--calFile` argument should be viewed as an optional argument entirely.  And any script that requires calibration instead by default make a DB query to request `cal_m` and `cal_b` in question.  The DB query should be made from the `vfatID`s in the input file, which can be obtained via:

```python
        # Determine vfatID
        import root_numpy as rp
        import numpy as np
        list_bNames = ['vfatN','vfatID']
        array_vfatData = rp.tree2array(tree=someGemTree, branches=list_bNames)
        array_vfatData = np.unique(array_vfatData)
``` 

Then `array_vfatData['vfatID']` is what is the starting point for the DB query.

If `--calFile` *is* provided, then the DB query is not made.

bdorney, 2019-03-07:
Superseded by https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/188

closed 2019-03-07 by bdorney