cms-gem-daq-project/gem-plotting-tools #155: Feature Request: DAC Scan Analysis Routine

bdorney, 2018-10-08:
## Brief summary of issue

We have a new DAC Scan routine for VFAT3 introduced in https://github.com/cms-gem-daq-project/vfatqc-python-scripts/pull/203.

Now we need a tool to analyze it.  

One critical thing to keep in mind is that `dacScanV3.py` can scan multiple optohybrids per call, so the output `TFile` may contain info for multiple detectors.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

Relevant pseudocode below:

```python
if __name__ == '__main__':
   setup an argparse.ArgumentParser instance
   get arguments from a user
   
   # Determine which links present suggest using root_numpy, something like:
   import root_numpy as rp
   import numpy as np
   bNames = ['link']
   ohArray = rp.tree2array(tree=calTree, branches=bNames)
   ohArray = np.unique(ohArray) # now ohArray will have one entry for each unmasked optohybrid

   # Determine which DAC was scanned and against which ADC
   nameX = ""
   nameY = ""
   for event in calTree:
      nameX = event.nameX
      nameY = event.nameY
      break # all entries will be the same

   # Get ADC0 or ADC1 calibration
   # nameY will either be ADC0 or ADC1, you'll need to load calibration for each link
   # use parseCalFile(...) from gem-plotting-tools/utils/anautilities.py
   # Note this calibration is unique to each (vfat,ohN) pairing
   # In future this is where a DB query based on vfatID would take place 

   # Make output plots, suggest TGraphErrors objects
   # I would suggest you use a nesteddict object (gempython/utils/nesteddict.py) to index these by [oh][vfat]
   for link in ohArray:
      for vfat in range(0,24):
           dict_dacScanResults[link][vfat] = r.TGraphErrors(...) # declare TGraphErrors
           You can get number of points in TGraphErrors by looking at maxVfat3DACSize of amc_user_functions_xhal.py (from cmsgemos PR#206)

   # Loop over events in the tree and fill plots
   for event in calTree:
       assign (dacValX, dacValY, dacValY_Err) to the tgraph for a given (vfat,oh) pair
       Here you need to use the ADCX (for X = 0,1) calibration to convert dacValY & dacValY_Err to physical units
       You also need to identify if nameX corresponds to a bias voltage DAC or a bias current DAC, see VFAT3 manual Table 25: GBL CFG CTR 4 : Monitoring settings
       Note nameX is the register that was scanned during the DAC scan, but we want plots that are Y=nameX (reg scanned) and X=nameY (ADC value), see below 
 
   # Fit the TGraphErrors Objects
   for oh in ohArray
      for vfat in range(0,24):
          dict_dacScanResults[oh][vfat].Fit(...) # linear fitting may not be suitable, polynomial may be required, investigation needed.  Possible try several fits and choose one with best norm chi2
   
    # Determine DAC values to achieve recommended bias voltage and current settings
    See Table 9, Nominal Current column, and the VFAT3a specific note under table 9
    See Table 10, Nominal Voltage column
    Here for each (vfat,oh) you should use the fit function to do a look-up on the nominal value to get the DAC value that will achieve this, e.g.
    dacVal = myTF1.Eval(nominal voltage)
    This then should be printed to terminal as a table, an output file (in either tab-delimited or TTree format) and then as a TGraph object with DAC value vs. VFAT position

    # Make plots
    Make our standard 3x8 grid plot of the DAC scan results, see make3x8Canvas(...), one such grid plot should be made for each entry in ohArray
    Store plots in an output TFile (use TDirectories to organize logically)
    
    Exit
```

You can take inspiration from [anaUltraScurve.py](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/anaUltraScurve.py).

Referencing the [vfat3 manual](https://espace.cern.ch/cms-project-GEMElectronics/VFAT3/VFAT3%20User%20Manual%20v2.2.pdf) may be necessary.

Several useful functions in the API:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/d7b78cef238894998fc350377c3e4ace1a0b4484/utils/anautilities.py#L341
https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/d7b78cef238894998fc350377c3e4ace1a0b4484/utils/anautilities.py#L438

## Context (for feature requests)

Need a tool for analyzing DAC scans.

bdorney, 2018-10-08:
@AndrewLevin

bdorney, 2018-10-08:
Looking for a volunteer to tackle this by next Monday.

AndrewLevin, 2018-10-08:
@bdorney, I will do it.

bdorney, 2018-10-10:
@AndrewLevin for those monitor (dac) select values that correspond to bias currents (See VFAT3 manual, Table 25 GBL CFG CTR 4 : Monitoring settings) you'll need to convert the ADC readback voltage back to a current.  Henri from VFAT3 team writes:

>The DACs are linear in the range we typically use them, but they tend to saturate on the higher end of the range. I have used 5th degree polynomial to do the fit, but I think lower order could be used. The currents from the DACs are converted to voltages for the ADCs with an external 20k Ohm resistor.

This also gives some hints as what type of fit functions could be applicable.  Recall the goal is to generate an automated routine.  So "tweaking" each fit is not an option, the fit should be chosen and design to always converge and give reliable results (like the scurve fit).

AndrewLevin, 2018-10-11:
@bdorney, I have two questions

1) regarding `assign (dacValX, dacValY, dacValY_Err) to the tgraph for a given (vfat,oh) pair`, a TGraph can be filled with errors in both the x and y directions, and I see that both errors are stored in the DAC scan (pending the merge of https://github.com/cms-gem-daq-project/vfatqc-python-scripts/pull/205)

2)  regarding `Note nameX is the register that was scanned during the DAC scan, but we want plots that are Y=nameX (reg scanned) and X=nameY (ADC value), see below`, I assume that nameX corresponds to dacValX, so then `assign (dacValX, dacValY, dacValY_Err) to the tgraph for a given (vfat,oh) pair` would create a TGraph with errors in only the x-direction, can you confirm that is what you intended?

bdorney, 2018-10-11:
Regarding question 1, see discussion here:
- https://mattermost.web.cern.ch/cms-gem-daq/pl/f3z8osj5s3nc9ejpynub59pcfa
- https://mattermost.web.cern.ch/cms-gem-daq/pl/mh334uwr7b8hpqsdctnjrxz6nw

Regarding question 2

```
   # Loop over events in the tree and fill plots
   for event in calTree:
       assign (dacValX, dacValY, dacValY_Err) to the tgraph for a given (vfat,oh) pair
       Here you need to use the ADCX (for X = 0,1) calibration to convert dacValY & dacValY_Err to physical units
       You also need to identify if nameX corresponds to a bias voltage DAC or a bias current DAC, see VFAT3 manual Table 25: GBL CFG CTR 4 : Monitoring settings
       Note nameX is the register that was scanned during the DAC scan, but we want plots that are Y=nameX (reg scanned) and X=nameY (ADC value), see below 
```

Yes `nameX` corresponds to `dacValX`, yes this is creating a `TGraphErrors` with only the x-direction having errors; the y-value represents the DAC.

e.g. what we want is that if we have a front end bias current that is supposed to be set to 100uA; we want to determine what that DAC value corresponds to, so it should be:

```
finalDACSetting = int(myFunc->Eval(100uA))
```

And this is how we use the dac scan information.  Is it clear?

AndrewLevin, 2018-10-12:
@bdorney, the script is written and working, except for these two parts:

```
   # Get ADC0 or ADC1 calibration
   # nameY will either be ADC0 or ADC1, you'll need to load calibration for each link
   # use parseCalFile(...) from gem-plotting-tools/utils/anautilities.py
   # Note this calibration is unique to each (vfat,ohN) pairing
   # In future this is where a DB query based on vfatID would take place 
```
```
    # Determine DAC values to achieve recommended bias voltage and current settings
    See Table 9, Nominal Current column, and the VFAT3a specific note under table 9
    See Table 10, Nominal Voltage column
    Here for each (vfat,oh) you should use the fit function to do a look-up on the nominal value to get the DAC value that will achieve this, e.g.
    dacVal = myTF1.Eval(nominal voltage)
    This then should be printed to terminal as a table, an output file (in either tab-delimited or TTree format) and then as a TGraph object with DAC value vs. VFAT position
```

Do you have either a calFile or further details about how to implement the calibration-related part?

Currently, I have a script that produces the 3 x 8 plots, but it is just based on the raw uncalibrated ADC and DAC values. Probably adding the calibration will not be difficult and the script should be ready and fully functional by Monday when you requested, but I do need some further information.

bdorney, 2018-11-13:
Closed by #158

closed 2018-11-13 by bdorney