cms-gem-daq-project/gem-plotting-tools #167: Error in anaDACScan.py

fsimone91, 2018-11-08:
Error while using analysis code

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior

```
(py2.7) [fsimone@gem904qc8daq GEB3c-l1-w]$ anaDACScan.py /data/bigdisk/GEM-Data-Taking/GE11_QC8/dacScanV3/2018.11.07.12.04/dacScanV3.root --calFileList calFileList_GEB3c-l1-w_ADC0.txt
Analyzing: '/data/bigdisk/GEM-Data-Taking/GE11_QC8/dacScanV3/2018.11.07.12.04/dacScanV3.root'
mkdir: cannot create directory ‘/data/bigdisk/GEM-Data-Taking/GE11_QC8//GE11-X-S-BARI-0010/dacScans/2018.11.07.12.04’: Permission denied
Traceback (most recent call last):
  File "/afs/cern.ch/user/f/fsimone/gemdaq/venv/cc7/py2.7/lib/python2.7/site-packages/gempython/scripts/anaDACScan.py", line 120, in <module>
    from utils.anaInfo import nominalDacValues
  File "/usr/lib64/python2.7/site-packages/ROOT.py", line 318, in _importhook
    return _orig_ihook( name, *args, **kwds )
ImportError: No module named utils.anaInfo
```
### Steps to Reproduce (for bugs)

1.
2.
3.
4.

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

* Version used:
* Shell used:

bdorney, 2018-11-08:
@AndrewLevin I can see you did not correctly test this inside a `venv` before submitting this PR.  This would have failed immediately from inside a `venv` due to the line:

```python
    from utils.anaInfo import nominalDacValues
```

Should definitely be something like:

```python
    from gempython.gemplotting.utils.anaInfo import nominalDacValues
```

In the future when submitting PR's please test them thoroughly.  This includes:

1. Ensuring the package builds properly, and 
2. Ensuring that after installing the package, either as an rpm or as a pip package in a `venv`, that the code works inside *that* env.

This impacts not only sysadmin, but as you see also users.

bdorney, 2018-11-08:
Other errors may exist...please resolve ASAP.

AndrewLevin, 2018-11-08:
Sorry about this. Here is the bugfix pull request: https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/168

commit 36f44db3 (2018-11-08): Merge pull request #168 from AndrewLevin/bugfix_anaDACScan_use_import_statements_that_work_in_venv resolves #167

in anaDACScan.py, use import statements that work in venv
closed 2018-11-08 by bdorney