cms-gem-daq-project/gem-plotting-tools #183: chamberInfo.py script currently on gem904daq01 machine

brentstone, 2019-02-19:
## Brief summary of issue
I am providing here a ```diff``` of the chamberInfo.py script that we are using on the ```gem904daq01``` machine (for use of GE2/1 currently) and the version of this script in the ```gem-plotting-tools``` repo on GitHub. See the ```diff``` below:

```bash
[gemuser@gem904daq01 gem-plotting-tools]$ diff mapping/chamberInfo.py /usr/lib/python2.7/site-packages/gempython/gemplotting/mapping/chamberInfo.py
16,29c16,18
<     #0:"GE11-VI-L-CERN-0002"
<     #Cosmic Stand
<     #0:"GE11-VI-L-CERN-0001"
<     #Point 5
<     0:"GEMINIm01L1",
<     1:"GEMINIm01L2",
<     2:"GEMINIm27L1",
<     3:"GEMINIm27L2",
<     4:"GEMINIm28L1",
<     5:"GEMINIm28L2",
<     6:"GEMINIm29L1",
<     7:"GEMINIm29L2",
<     8:"GEMINIm30L1",
<     9:"GEMINIm30L2"
---
>     0:"GE11-X-S-INDIA-0002",
>     1:"GE21-M1-P4",
>     2:"GE11-X-S-NOCOPPER-0001"
34,37d22
<     #0:"long"
<     #Cosmic Stand
<     #0:"long"
<     #Point 5
40,47c25
<     2:"short",
<     3:"short",
<     4:"long",
<     5:"long",
<     6:"short",
<     7:"short",
<     8:"long",
<     9:"long"
---
>     2:"short"
52,55d29
<     #0:0x0
<     #Cosmic Stand
<     #0:0xF40400
<     #Point 5
57,65c31,32
<     1:0x0,
<     2:0x0,
<     3:0x0,
<     4:0x0,
<     5:0x0,
<     6:0x0,
<     7:0x0,
<     8:0x0,
<     9:0x0
---
>     1:0xfff000,
>     2:0x1
96,172c63,146
<     #    0:{
<     #        #Pulse Stertch
<     #        "CFG_PULSE_STRETCH":4,
<     #        #Ensure the cal pulse is off
<     #        "CFG_CAL_MODE":0,
<     #        #Set the Latency - 10x10 PMT on R&D Setup
<     #        "CFG_LATENCY":98,
<     #        #Correct the bug in the shaper
<     #        #"CFG_PT":0xf,
<     #        "CFG_PT":3,
<     #        #VFAT3a DAC settings
<     #        #"CFG_BIAS_SH_I_BDIFF":150,
<     #        #"CFG_BIAS_SH_I_BFCAS":250,
<     #        #"CFG_BIAS_SD_I_BDIFF":255,
<     #        #"CFG_BIAS_SD_I_BFCAS":255,
<     #        #VFAT3b DAC settings
<     #        "CFG_BIAS_SH_I_BDIFF":80,
<     #        "CFG_BIAS_SH_I_BFCAS":130,
<     #        "CFG_BIAS_SD_I_BDIFF":140,
<     #        "CFG_BIAS_SD_I_BFCAS":135,
<     #        #Provide a slight offset to the ZCC comparator baseline voltage
<     #        "CFG_THR_ZCC_DAC":10,
<     #        #High VFAT3 preamp gain
<     #        #"CFG_RES_PRE":1,
<     #        #"CFG_CAP_PRE":0,
<     #        #Medium VFAT3 preamp gain
<     #        "CFG_RES_PRE":2,
<     #        "CFG_CAP_PRE":1,
<     #        #Low VFAT3 preamp gain
<     #        #"CFG_RES_PRE":4,
<     #        #"CFG_CAP_PRE":3,
<     #        #Comparator Mode - CFD
<     #        "CFG_SEL_COMP_MODE":0,
<     #        "CFG_FORCE_EN_ZCC":0
<     #        #Comparator Mode - ARM
<     #        #"CFG_SEL_COMP_MODE":1,
<     #        #"CFG_FORCE_EN_ZCC":0
<     #        #Comparator Mode - ZCC
<     #        #"CFG_SEL_COMP_MODE":2,
<     #        #"CFG_FORCE_EN_ZCC":1
<     #        },
<     #    1:{
<     #        #Pulse Stertch
<     #        "CFG_PULSE_STRETCH":4,
<     #        #Ensure the cal pulse is off
<     #        "CFG_CAL_MODE":0,
<     #        #Set the Latency - 10x10 PMT on R&D Setup
<     #        "CFG_LATENCY":98,
<     #        #Correct the bug in the shaper
<     #        #"CFG_PT":0xf,
<     #        "CFG_PT":3,
<     #        #Updated DAC settings from Flavio
<     #        "CFG_BIAS_SH_I_BDIFF":150,
<     #        "CFG_BIAS_SH_I_BFCAS":250,
<     #        "CFG_BIAS_SD_I_BDIFF":255,
<     #        "CFG_BIAS_SD_I_BFCAS":255,
<     #        #Provide a slight offset to the ZCC comparator baseline voltage
<     #        "CFG_THR_ZCC_DAC":10,
<     #        #High VFAT3 preamp gain
<     #        #"CFG_RES_PRE":1,
<     #        #"CFG_CAP_PRE":0,
<     #        #Medium VFAT3 preamp gain
<     #        "CFG_RES_PRE":2,
<     #        "CFG_CAP_PRE":1,
<     #        #Low VFAT3 preamp gain
<     #        #"CFG_RES_PRE":4,
<     #        #"CFG_CAP_PRE":3,
<     #        #Comparator Mode - CFD
<     #        "CFG_SEL_COMP_MODE":0,
<     #        "CFG_FORCE_EN_ZCC":0
<     #        #Comparator Mode - ARM
<     #        #"CFG_SEL_COMP_MODE":1,
<     #        #"CFG_FORCE_EN_ZCC":0
<     #        #Comparator Mode - ZCC
<     #        #"CFG_SEL_COMP_MODE":2,
<     #        #"CFG_FORCE_EN_ZCC":1
<     #        }
---
>         0:{
>             #Pulse Stertch
>             "CFG_PULSE_STRETCH":3,
>             #Ensure the cal pulse is off
>             "CFG_CAL_MODE":0,
>             #Set the Latency - 10x10 PMT on R&D Setup
>             "CFG_LATENCY":142,
>             #Correct shaper for comp mode
>             "CFG_PT":0xf,
>             "CFG_FP_FE":0x7,
>             #Provide a slight offset to the ZCC comparator baseline voltage
>             "CFG_THR_ZCC_DAC":10,
>             #High VFAT3 preamp gain
>             #"CFG_RES_PRE":1,
>             #"CFG_CAP_PRE":0,
>             #Medium VFAT3 preamp gain
>             "CFG_RES_PRE":2,
>             "CFG_CAP_PRE":1,
>             #Low VFAT3 preamp gain
>             #"CFG_RES_PRE":4,
>             #"CFG_CAP_PRE":3,
>             #Comparator Mode - CFD
>             "CFG_SEL_COMP_MODE":0,
>             "CFG_FORCE_EN_ZCC":0
>             #Comparator Mode - ARM
>             #"CFG_SEL_COMP_MODE":1,
>             #"CFG_FORCE_EN_ZCC":0
>             },
>         1:{
>             #Pulse Stertch
>             "CFG_PULSE_STRETCH":3,
>             #Ensure the cal pulse is off
>             "CFG_CAL_MODE":0,
>             #Set the Latency - 10x10 PMT on R&D Setup
>             "CFG_LATENCY":142,
>             #Correct shaper for comp mode
>             "CFG_PT":0xf,
>             "CFG_FP_FE":0x7,
>             #Provide a slight offset to the ZCC comparator baseline voltage
>             "CFG_THR_ZCC_DAC":10,
>             #High VFAT3 preamp gain
>             #"CFG_RES_PRE":1,
>             #"CFG_CAP_PRE":0,
>             #Medium VFAT3 preamp gain
>             "CFG_RES_PRE":2,
>             "CFG_CAP_PRE":1,
>             #Low VFAT3 preamp gain
>             #"CFG_RES_PRE":4,
>             #"CFG_CAP_PRE":3,
>             #Comparator Mode - CFD
>             "CFG_SEL_COMP_MODE":0,
>             "CFG_FORCE_EN_ZCC":0
>             #Comparator Mode - ARM
>             #"CFG_SEL_COMP_MODE":1,
>             #"CFG_FORCE_EN_ZCC":0
>             },
>         2:{
>             #Pulse Stertch
>             "CFG_PULSE_STRETCH":3,
>             #Ensure the cal pulse is off
>             "CFG_CAL_MODE":0,
>             #Set the Latency - 10x10 PMT on R&D Setup
>             "CFG_LATENCY":142,
>             #Correct shaper for comp mode
>             "CFG_PT":0xf,
>             "CFG_FP_FE":0x7,
>             #Provide a slight offset to the ZCC comparator baseline voltage
>             "CFG_THR_ZCC_DAC":10,
>             #High VFAT3 preamp gain
>             #"CFG_RES_PRE":1,
>             #"CFG_CAP_PRE":0,
>             #Medium VFAT3 preamp gain
>             "CFG_RES_PRE":2,
>             "CFG_CAP_PRE":1,
>             #Low VFAT3 preamp gain
>             #"CFG_RES_PRE":4,
>             #"CFG_CAP_PRE":3,
>             #Comparator Mode - CFD
>             "CFG_SEL_COMP_MODE":0,
>             "CFG_FORCE_EN_ZCC":0
>             #Comparator Mode - ARM
>             #"CFG_SEL_COMP_MODE":1,
>             #"CFG_FORCE_EN_ZCC":0
>             }
```
### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior

### Steps to Reproduce (for bugs)

1.
2.
3.
4.

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

* Version used:
* Shell used:

bdorney, 2019-02-19:
The version in the central repo should *always* match the P5 system.

The version on a test stand is the responsible of the test stand admin.

This is why the `editConfig` alias is provided for you on the `gemuser` account of both `gem904qc8daq` and `gem904daq01`.  If the test stand configuration has changed then you need to use the `editConfig` alias to update the corresponding dictionaries.

Note the system installed version on the machine is a system level symlink that points to a user editable area (which the `editConfig` alias allows you to edit).

Closing as "feature not a bug won't fix"

closed 2019-02-19 by bdorney

brentstone, 2019-02-19:
@bdorney, I was told by @mexanick to open an issue for this, so just letting @mexanick know that this has been closed by @bdorney