cms-gem-daq-project/gem-plotting-tools #184: Update `calibrateThrDac.py` to use nonliner fit function

bdorney, 2019-02-22:
## Brief summary of issue

Linear fit applied for calibration of `CFG_THR_ARM_DAC` should be dropped in favor of a nonlinear fit function.

The DAC itself is nonlinear at low DAC values and also at high DAC values.  High DAC values are not the concern, however if the fit curve doesn't accurately describe the data the trimming procedure will fail to align the channels.

See discussion on [slides 11 & 16](https://indico.cern.ch/event/792165/contributions/3290305/attachments/1784793/2905345/BDorney_GEMDAQMtg_20180124_xtalk.pdf)

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

Fit function should accurately describe the data, especially at the DAC values where the DAC is most useable.

## Current Behavior

Fit function only describes the DAC in the linear range.

## Context (for feature requests)

Linear fit of the dac calibration causes the trim algorithm to fail if a particular VFAT should be trimmed in this region (e.g. noise is low enough for DAC to be placed in this region).

## Your Environment

* Version used: b96aa99d6c48ec0e26a1593e30400d558251b0ba

bdorney, 2019-02-22:
@AndrewLevin do you think you could tackle this by next Wednesday?

AndrewLevin, 2019-02-22:
Yes, I will do it by next Wednesday.

AndrewLevin, 2019-02-23:
Can you point me to some data to test the updated fitting function?

bdorney, 2019-02-24:
See:

https://github.com/cms-gem-daq-project/sw_utils/issues/18

bdorney, 2019-02-24:
Don’t overwrite the existing data if you can avoid it.

AndrewLevin, 2019-02-26:
With a fifth order polynomial I get

[image]

instead of

[image]

Does it look good?

bdorney, 2019-02-26:
But how does it look at low values of ARM DAC? e.g. in range 20 to 80.  The high range is the less useful range and we wouldn't really operate there (cuts the signal).  

Can you show a zoom in the requested region; also a data/fit would be interesting to see.

AndrewLevin, 2019-02-26:
In the data that you provided in https://github.com/cms-gem-daq-project/sw_utils/issues/18, there is basically no nonlinearity in the region 20 to 80. Here is the zoom of the data and a linear fit in that region:
[image]

bdorney, 2019-02-26:
Why don't you try to investigate further with the other recent measurements.  See for example the cases returned by:

```bash
ls $DATA_PATH/*/armDacCal/current
```

bdorney, 2019-02-26:
Also it would be good if the algorithm could try to identify when it should *stop* fitting; and use a *per* VFAT fit range rather than a common fit range applied to all hybrids on a link. 

The algorithm should *stop* fitting when the ARM DAC setting puts the comparator such that it is sensitive to the pedestal (you can see this in the scurve mean vs. arm DAC when at low arm DAC in some of the above plots the scurve mean point jumps up).  To determine when it should stop fitting as a first approach I think it should look when the ENC (e.g. scurve sigma) starts to blow up, e.g. deviate significantly from the high ARM DAC ENC value; this would be a good place to stop.  There might be a better approach; I'm interested in what you find.

AndrewLevin, 2019-02-26:
There are 6 listOfScanDates files, two of them contain scandates that do not exist (`$DATA_PATH//GE11-X-S-CERN-0009/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-CERN-0010.txt` and `$DATA_PATH//GE11-X-S-CERN-0010/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-CERN-0009.txt`), one of them (`$DATA_PATH//GE11-X-L-CERN-0002/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-L-CERN-0002.txt`) seems to have a data format problem:

```
Info in ROOT::Math::ParameterSettings>: lower/upper bounds outside current parameter value. The value will be set to (low+up)/2 
Info in ROOT::Math::ParameterSettings>: lower/upper bounds outside current parameter value. The value will be set to (low+up)/2 
Traceback (most recent call last):
  File "calibrateThrDac.py", line 229, in <module>
    mapVFATPos2VFATSN[vfat]['serialNum'] = array_vfatData[array_vfatData['vfatN'] == vfat]['vfatID']
TypeError: Cannot cast array data from dtype('float32') to dtype('uint32') according to the rule 'safe'
```
and the other three (`$DATA_PATH//GE11-X-S-CERN-0007/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-CERN-0007.txt`, `$DATA_PATH//GE11-X-S-BARI-0006/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-BARI-0006.txt`, and `$DATA_PATH//GE11-X-S-INDIA-0007/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-INDIA-0007.txt` seem to have basically negligible nonlinearities especially if you take the error bars seriously:

[image]

[image]

[image]

bdorney, 2019-02-26:
> one of them (`$DATA_PATH//GE11-X-L-CERN-0002/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-L-CERN-0002.txt`) seems to have a data format problem:

Issue is that bit flips occur when the chipID is requested from the VFAT and then is beyond what an integer of type `long` can hold which is all the [array class](https://docs.python.org/2/library/array.html) can hold and so a hot fix was applied to change it to a float.  You'll have to force it during debugging sessions.  Hopefully the Reed-Muller encoded hybrids will make this point mute. 

> and the other three (`$DATA_PATH//GE11-X-S-CERN-0007/armDacCal/current/listOfScanDates_calibrateArmDac_GE11-X-S-CERN-0007.txt`, [...]

See the PM I sent you for another trial.  The range of nonlinear is different for *each* hybrid and the issue here is the fringe cases.  

Also how's the automatic determination of fit range coming?

AndrewLevin, 2019-02-27:
I just made a pull request to address this issue: https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/186

commit a1012352 (2019-04-19): Merge pull request #186 from AndrewLevin/for_issue_184

Updates to `calibrateThrDac.py` fits
* removal of low `thrDac` values sensitive to pedestal 
* use of non-linear function 

Closes #184
closed 2019-04-19 by jsturdy