cms-gem-daq-project/gem-plotting-tools #197: Bug report: anaDACScan.py overwrites .png output files

ElizabethRoseStarling, 2019-04-04:
## Brief summary of issue
When utilizing the anaDACScan.py script, the summary .png files which are outputted are all given the same name, so that after the script finishes running, there is only one image file saved instead of ~16.

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

Each .png summary image which is outputted should have a unique name, indicating the DAC value with which it is correlated.

## Current Behavior

Currently each image is given the same name, with nothing to differentiate it from the others, so they overwrite each other and only the last outputted image is saved in the directory.

### Steps to Reproduce (for bugs)

1. anaDACScan.py --calFileList /home/gemuser/CalFileList.txt /data/bigdisk/GEM-Data-Taking/SliceTest/dacScanV3/2019.04.03.16.02/dacScanV3.root

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

* Version used: current version located in gemuser@gem904daq01
* Screen used: LizScans

bdorney, 2019-04-04:
@fsimone91 has volunteered to take on this bug.

bdorney, 2019-04-05:
Addressed by #198 thanks @fsimone91

closed 2019-04-05 by bdorney