cms-gem-daq-project/gem-plotting-tools #203: Bug Report: Nominal DAC Values for DAC Scan Analysis Incorrect

bdorney, 2019-05-21:
## Brief summary of issue

Turns out some of the nominal values reported here:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/1fdf018d8f965383a05955c93d9422f9beb383f9/utils/anaInfo.py#L17-L34

Does not match the values reported in Tables 12 & 13 of the [VFAT3 Manual V2.2](https://espace.cern.ch/cms-project-GEMElectronics/VFAT3/VFAT3%20User%20Manual%20v2.2.pdf).

### Specific Concern for `CFG_HYST`
In a private communication the VFAT3 team stated also that the `CFG_HYST` register should  be kept at a value below `0xA`.  We should add a constraint on this register.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

The DAC's below need to have their nominal values updated.

Additionally we should set a constraint on the `CFG_HYST` register to provide values always below `0xA`.

## Current Behavior

Specifically the following DAC's do not match the value looked in VFAT3 manual:

| Register | Val in GEM SW | Val in VFAT3 Man. |
| :------- | :---------------: | :----------------: |
| CFG_BIAS_PRE_I_BIT | 150 uA | 30 uA |
| CFG_BIAS_PRE_I_BLCC | 25 nA | 2.5 uA |
| CFG_BIAS_PRE_I_BSF | 26 uA | 6.5 uA |
| CFG_BIAS_SD_I_BSF | 30 uA | 7.5 uA |
| CFG_HYST | 100 nA | 500 nA |
| CFG_THR_ZCC_DAC | 5.5 mV | 22 mV |

## Context (for feature requests)

Our DAC scan analysis is not assigning the correct DAC value to the above registers.  We are not using the chip correctly. 

## Your Environment

* Version used: 1fdf018d8f965383a05955c93d9422f9beb383f9

bdorney, 2019-05-21:
I did a double check of the scaling factors, e.g.:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/1fdf018d8f965383a05955c93d9422f9beb383f9/utils/anaInfo.py#L36-L54

And they are all correct.

AndrewLevin, 2019-05-21:
The values that are there now are the "Nominal Current" in Table 9 and the "Nominal Voltage" in Table 10.

bdorney, 2019-05-21:
> The values that are there now are the "Nominal Current" in Table 9 and the "Nominal Voltage" in Table 10.

Looks like we'll have to ask the VFAT3 team what the difference is...

bdorney, 2019-05-23:
VFAT3 team reports that Tables 9 and 10 should be used for configuring and data taking.

Closing as "NotABugWontFix"

closed 2019-05-23 by bdorney