cms-gem-daq-project/gem-plotting-tools #208: in calibrateThrDac.py, Gaussian fits can fail

AndrewLevin, 2019-05-22:
The Gaussian fit to the scurveMean distribution can fail, and there is currently no protection against this.

## Brief summary of issue
In calibrateThrDac.py, a Gaussian is fitted to the per-vfat scurveMean distribution in order to determine the mean of the scurveMean distribution and the width of the scurveMean distribution. The Gaussian fit can fail, and it fails visibly in at least the 7 cases below.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Possible Solution (for bugs)
I would propose that we remove points that satisfy scurveMeanError/scurveMean < 0.001. Also, we may want to look into how these histograms/tgraphs are binned in the scurve analysis script.

## Context (for feature requests)
In VFAT18 in the plot below, the error on the highest THR_DAC point for VFAT18 is around 6*10^-7, which causes the fit to visibly fail because that point is weighted to highly.

[image]

The solution above removes this point, and then the fit looks reasonable.
[image]

## Your Environment

* Version used:
* Shell used:

bdorney, 2019-05-23:
In this particular case why is the fit range not modified? e.g. it seems to still be trying to include that part of the DAC.  Isn't the fit range automatically determined?  The polynomial seems a bit "bendy" as a result of trying to use this range.

Additionally what about the failures reported [here](https://indico.cern.ch/event/822091/contributions/3436929/attachments/1847158/3031036/fsimone_QC8PFA_20052019.pdf).  Ignoring VFATs where the scurve data was actually bad, e.g. slide 4 VFAT7, there are cases where a single point is weighting the fitting; or there are cases where the scurve mean vs. ARM DAC curve is completely missing the low ARM DAC range; this is problematic and we should try to address this.

AndrewLevin, 2019-05-23:
My proposal is to remove the point, which is equivalent to modifying the fit range.

This issue is not related to the failures reported in Federica's slides.

 I will create a separate issue for that shortly.

bdorney, 2019-05-23:
>My proposal is to remove the point, which is equivalent to modifying the fit range.

I'm not sure it is the same thing.  In your example above you have removed the point, but the fit is still allowed to be in that region (see above).  While I agree we should remove the point, in this case since the point is at the edge of the range we should also reduce the range.

 > I can create a separate issue for that, or should I consolidate both issues?

Separate issue is a good idea.

AndrewLevin, 2019-05-23:
Yes, the fit range could be automatically set to the minimum and the maximum unremoved points. That would not affect the fitted coefficients, but would affect the plots.

Here is the separate issue: https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/210

bdorney, 2019-05-23:
> Yes, the fit range could be automatically set to the minimum and the maximum unremoved points. That would not affect the fitted coefficients, but would affect the plots.

Let's do that.

commit f0e28441 (2019-05-28): Merge pull request #212 from AndrewLevin/update_calibratethrdac

fixes #208

* in `calibrateThrDac.py`, remove points with low `scurveMeanError` and restrict fit range
closed 2019-05-28 by jsturdy