cms-gem-daq-project/gem-plotting-tools #210: poor scurveMean vs thrDac fits reported by QC8 PFA analyzers

AndrewLevin, 2019-05-23:
Poorly fitted scurveMean vs thrDac data found by the QC8 team indicate further changes to the fit algorithm are necessary.

## Brief summary of issue
In slides 7 and 16 of https://indico.cern.ch/event/822091/contributions/3436929/attachments/1847158/3031036/fsimone_QC8PFA_20052019.pdf multiple examples can been seen where the fit is not properly modeling the non-linear behavior at low thrDac.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior
Fit should be very sensitive to low thrDac points. Even individual one-off points that should pull the fit down.

## Current Behavior
Fit often looks linear in the low thrDac region when the data does not.

## Context (for feature requests)
Trimming is useless when the low thrDac region is modeled this badly.

## Your Environment

* Version used:
* Shell used:

AndrewLevin, 2019-05-28:
One possibly solution would be to only fit the region THR_DAC < 100, which would reduce the number of wiggles that the polynomial must model. From Marek's plots

https://mattermost.web.cern.ch/api/v4/files/izr38nebhbbbfpj7r86msaj8ao?download=1

it seems we are always trimming at THR_DAC < 60, so probably we do not need to fit the region THR_DAC > 100.

bdorney, 2019-06-03:
> One possible solution would be to only fit the region THR_DAC < 100, which would reduce the number of wiggles that the polynomial must model. From Marek's plots
> 
> https://mattermost.web.cern.ch/api/v4/files/izr38nebhbbbfpj7r86msaj8ao?download=1
> 
> it seems we are always trimming at THR_DAC < 60, so probably we do not need to fit the region THR_DAC > 100.

Did you get the data and try to apply this proposal to the plot above? Any updates?

bdorney, 2019-08-01:
potentially addressed by #239; will re-open should this persist afterward.

closed 2019-08-01 by bdorney