cms-gem-daq-project/gem-plotting-tools #214: Issues in Virtual Environment

bregnery, 2019-06-05:
Getting pip to install the correct packages in the virtual environment results in conflicts.

### Types of issue
- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior
```bash
$ pip install -U requirements-dev.txt
$ pip install rpm/gempython_gemplotting-1.3.1-dev1.dev45.zip
```
Then, the package is ready to use in the vitual environment

## Current Behavior
```bash
$ pip install -U requirements-dev.txt

It looks like you meant to type `pip install -r requirements-dev.txt`, but you left out the `-r` by accident. Aborting installation.

$ pip install -r requirements-dev.txt

Exception:
Traceback (most recent call last):
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pip/basecommand.py", line 134, in main
    status = self.run(options, args)
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pip/commands/install.py", line 220, in run
    for req in parse_requirements(filename, finder=finder, options=options):
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pip/req.py", line 1477, in parse_requirements
    req = InstallRequirement.from_line(line, comes_from, prereleases=getattr(options, "pre", None))
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pip/req.py", line 129, in from_line
    return cls(req, comes_from, url=url, prereleases=prereleases)
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pip/req.py", line 44, in __init__
    req = pkg_resources.Requirement.parse(req)
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pkg_resources.py", line 2914, in parse
    reqs = list(parse_requirements(s))
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pkg_resources.py", line 2839, in parse_requirements
    line, p, specs = scan_list(VERSION,LINE_END,line,p,(1,2),"version spec")
  File "/data/bigdisk/users/bregnery/venvs/inflectionEnv/lib/python2.7/site-packages/pkg_resources.py", line 2817, in scan_list
    "Expected ',' or end-of-list in",line,"at",line[p:]
ValueError: ("Expected ',' or end-of-list in", "sphinx<1.5; python_version == '2.6',", 'at', "; python_version == '2.6',")
```

### Steps to Reproduce (for bugs)
Follow commands in current behavior

## Possible Solution (for bugs)
Update `requirements-dev.txt` to be used with the commands contained in `.ci/test_on_docker.sh`
```bash
$    pip install -U "pip<10" importlib
$   pip install -U setuptools
$    pip install -U codecov
$    pip install -U -r requirements.txt
$   pip install -U root_numpy
$   pip install rpm/gempython_gemplotting-1.3.1-dev1.dev45.zip
```
This correctly installs the package in the virtual environment

## Context (for feature requests)
Trying to install package in virtual environment

## Your Environment
* Version used: f0e2844171cafd9184b2a6d1ba639523d2fee19b
* Shell used: Bash

bdorney, 2019-06-05:
Instead of trying to install the requirements and then the package did you try to just install the package directly? e.g. just:

```
$ pip install rpm/gempython_gemplotting-1.3.1-dev1.dev45.zip
```

Here `pip` should look at the package, see what it requires, attempt to automatically install the requirements and then install the package.  There might be an issue in the `requirements` txt file but I expect installing the package itself would work.

bregnery, 2019-06-05:
I tried that and it failed. I don't have the error message. I could try again tomorrow with a new virtual environment and put the error message here.

jsturdy, 2019-06-05:
Sorry, two separate issues here:
* `pip install -U -r requirements-dev.txt` should be used in the developing environment (these packages are required for running the build process)
* `pip install -U -r requirements.txt` should be run in the testing environment

These two steps are what the `.ci/test_on_docker.sh` script is basically doing

These environments can be the same

The second issue is that `requirements-dev.txt` seems broken currently, but should double check with the correct syntax (using `-U` for upgrade, in addition to `-r` to specify the requirements file)