cms-gem-daq-project/gem-plotting-tools #226: Feature Request: GE2/1 Compatibility for Plotting

sbutalla, 2019-06-25:
## Brief summary of issue

Currently, the `anaUltraScurve.py` and related scripts are compatible with GE1/1 only. The GE2/1 has only 12 VFATs per module, while the GE1/1 has 24, so half of the plots are not populated and can therefore be discarded when plotting the final result for GE2/1 data.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

In order to change this, import the hardware constants from `gempython.tools`.
The user can add an option specifying the `--gemType`, being either the GE11 or GE21 (ME0 to be implemented later), as well as the `detType` argument (i.e., long/short for GE11, module number for GE21). After this, the number of VFATs would be assigned to a variable which is used to set the maximum of each of the loops that controls plotting for the individual VFATs.

## Current Behavior

The analysis script will create plots for the correct number of VFATs depending on the GEM detector, which will be provided as an option. I.e., for the GE2/1 plot only 12 VFATs per module, and for the GE1/1 plot 24 VFATs.

## Possible Solution (for bugs)

Retrieve the hardware constants from `hw_constants.py`, add option for `gemVariant` and use the `vfatsPerGemVariant` to assign the number of VFATs for the proper `gemType`, and set the default value of `gemType` to GE11 to maintain backwards compatibility:
```
from gempython.tools.hw_constants import gemVariants, vfatsPerGemVariant
parser.add_option("--gemType", dest="gemType", type="string", default="ge11", help="String that defines the GEM variant, available from the list: {0}".format(gemVariants.keys()))
parser.add_option("--detType",type="string",help="Detector type within gemType. If gemType is 'ge11' then this should be from list {0}; if gemType is 'ge21' then this should be from list {1}; and if type is 'me0' then this should be from the list {2}".format(gemVariants['ge11'],gemVariants['ge21'],gemVariants['me0']),default=None)

#Assign number of VFATs for gemType
    if args.gemType == "ge11":
        vfatNum = {0}.format(vfatsPerGemVariant)
    elif args.gemType == "ge21":
        vfatNum = {0}.format(vfatsPerGemVariant)
    else:
        print("me0 gemType not currently implemented, exiting.")
        return
```
For every loop that is plotting distributions, use the maximum range of the loop as `vfatNum`, i.e.,

`for vfat in range(0,vfatNum):`

## Your Environment

* Version used: `135f4d8`
* Shell used: `bash`

bdorney, 2019-06-25:
The plotting tools should not require any additional arguments to handle the different detector types. 
 The better solution is to add additional metadata into the raw data files (defined in `vfatqc-python-scripts/utils/treeStructure.py`), e.g. `TBranches` called `gemType` and `detType` that contain string data.

Then at the start of each routine this should be read from from the `TTree` automatically and the routines should determine the index range to use, container size, etc...

Additionally more than *just* the loop variables would need to change.  We would need GE2/1 versions of the following dictionaries:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/mapping/chamberInfo.py#L37-L56

And

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/mapping/chamberInfo.py#L97-L107

Then on the fly the plotting routines should determine whether they should use [`make3x8Canvas`](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/utils/anautilities.py#L998-L1026), [`make2x4Canvas`](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/utils/anautilities.py#L969-L996) or their GE2/1 equivalents.

This can probably be accomplished by changing all instances of `make3x8Canvas` and `make2x4Canvas` to [`saveSummary`](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/utils/anautilities.py#L1305-L1359) and [`saveSummaryByiEta`](https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/utils/anautilities.py#L1361-L1394) but this may not be appropriate as in many cases the `makeAxBCanvas` functions return a TCanvas object while the `saveSummary*` functions do not return any object.

Care must be taken and the proposal should be well thought out and *clear* outlined before development begins or a PR is submitted.

bdorney, 2019-06-25:
Additionally many distributions are plotted as summaries that have x axis as VFAT number so this would need to be updated as well; containers sizes, etc...

dteague, 2019-07-12:
# Plan for changes
## Overview
This change is to follow the basic outline given by Brian. 

First, a new metadata to the files called ```gemType``` and ```detType``` that contain the basic info to be propagated into the plotting software. A helper function (put in ```anautilities.py```) can be called to set these variables at the beginning of every file. To keep this change as non-destructive as possible, if these TBranches aren't found, they default to "ge11". The geometry info will continue to be stored in ```mapping/chamberInfo.py```, so the code can grab values from this file such as number of iEta, iPhi, VFATs, etc.

For the canvases, the plot can have it's properties set based on the gem and detector type (ie 3x8 plot vs 3x4 plot for ge11 and ge21), so the make#x#Canvas can be absorbed into the saveSummary which would return the canvas and have an option for saving the Summary (possible renaming needed then!) 

## Files to be changed
+ **chamberInfo.py:**
File would be expanded to have dictionaries and those dictionaries set by some setting function called after the import. The actual maps needed by the code, ```chamber_vfatPos2iEta```, ```chamber_vfatPos2iEtaiPhi```, ```chamber_vfatPos2PadIdx```, ```chamber_iEta2VFATPos``` will automatically be set to "ge11" specifications so if the setting function is forgotten, it will still default to the old version

+ **anautilities.py**
A function containing of map for grabbing the number of vfats, iphi, ieta, etc can be created here to remove the magic numbers (namely 24) from code. If the function finds that the branches it needs don't exist, it defaults to the "ge11" specifications

  Also, the make#x#Canvas functions will be removed and their functionality expanded and put into the saveSummary function which will always return the TCanvas and have a toggle for saving the canvas. This will mean all references to these functions will have to change, but it should be fairly trivial since functionality will be roughly the same save for a slight change in input variables.

+ **ana scripts** 
All will have to load the gem and detector type and ancillary variables and put them either into global variables or add them as arguments to functions that require them. Most all of the changes in these files are removing magic numbers and replacing them with the values used for each gem and detector type.

## Summary
With these changes, it will it should keep functionality of all programs in the ```ana_scan.py``` suite completely the same save for functions having different names. This will require a change in the ```vfatqc-python-scripts```, so another issue/PR will need to be made in regards to this, but by putting default values if these branches aren't found, it can be backwards compatible. Following these specifications, it should cause no change for the user and should automatically update plots depending on the gem and detector.

 ----

Let me know if there are things I should expand on and/or problems that might need to be fixed with this plan!

bdorney, 2019-07-12:
> File would be expanded to have dictionaries and those dictionaries set by some setting function called after the import. The actual maps needed by the code, chamber_vfatPos2iEta, [...]

These are dictionaries; so why aren't we just adding additional outer keys, e.g.

```
chamber_vfatPos2iEta["ge11"] = nested dictionary
chamber_vfatPos2iEta["ge21"] = nested dictionary
```

e.g. `chamber_vfatPos2iEta["ge11"]` defaults to what we have now.  Same idea for all other dictionaries.

Question: do you expect M1...M8 to have *different* VFAT layout? e.g. is VFAT0 always in the same general location on all detector types? If so then we do not need a `detType` inner key to `gemType` in these dictionaries.

> A function containing of map for grabbing the number of vfats, iphi, ieta, etc can be created here to remove the magic numbers (namely 24) from code. If the function finds that the branches it needs don't exist, it defaults to the "ge11" specifications

Why is a function needed?  If `gemType` and `detType` are stored as metadata in the raw data then the dictionaries from [`hw_constants.py`](https://github.com/cms-gem-daq-project/cmsgemos/blob/generic-amc-RPC-v3-short-term/gempython/tools/hw_constants.py) or their `boost`ed equivalents from C++ cmsgemos header files can be used directly, .e.g.

```python
for vfat in range(vfatsPerGemVariant[gemType]):
   do something
```

This replaces all for loop indices; or it's C++ `boost`ed equivalent from `cmsgemos`.

> Also, the make#x#Canvas functions will be removed and their functionality expanded and put into the saveSummary function which will always return the TCanvas and have a toggle for saving the canvas. [...]

It would be good to see your proposed prototype *before* submitting a PR and how to refactor various algorithms to use this.

> All will have to load the gem and detector type and ancillary variables and put them either into global variables or add them as arguments to functions that require them. Most all of the changes in [...]

It seems like this is not really thoroughly thought through.  Again it would be good to see your proposed prototypes before a PR is submitted.

 ----

Seems like the proposals here would be API breaking proposals and I think @jsturdy would like a certain direction to be taken here and a timeline for this to be established; I differ to him on this.

dteague, 2019-07-31:
# Plan Revision

## Basic Changes 
Here are the basic changes that can be made to the code such that there is no change in the overall code base at all:

### chamberInfo.py
The chambers are generalized for all gemTypes available ("ge11", "ge21", "me0"). All the chamber dictionaries (`chamber_vfatPos2PadIdx`, `chamber_vfatPos2iEta`, `chamber_vfatPos2iEtaiPhi`, `chamber_iEta2VFATPos`) are made into larger dictionaries with the first argument being the gemType, ie

```python
chamber_vfatPos2PadIdx[gemType] = chamber_vfatPos2PadIdx<Old version>
```

To aid in readability, this could be renamed to be something more descriptive like "`chamber_gem2vfatPos2PadIdx`" or something. This gets clunky very quickly, so suggestions are welcome!! Right now, the naming convention is kept the same as the non-gemType-dictionaried map. As a side note: all the dictionaries are derived from the `chamber_vfatPos2PadIdx` as in the old code.

These 4 dictionaries are used in the following spots in the code:

* `scurveAlgos.py`
* `anaSBitReadout.py`
* `anautilities.py`: `get2DMapOfDetector()`
* `anautilities.py`: `make3x8Canvas()`
* `anautilities.py`: `saveSummary()`

So these will have to be updated most immediately to propagate the `chamberInfo.py` changes.

Also, the max iEta and iPhi numbers are needed for setting up the canvases as well as looping correctly. This could be found by getting this from the length of `chamber_iEta2VFATPos` dictionary, but to aid in readability, a new dictionary was created in `chamberInfo.py` called `chamber_maxiEtaiPhiPair` which is of the format:
```python
chamber_maxiEtaiPhiPair = {"ge11": (8,3),
	                       "ge21": (4,3), etc.}
```

but this could also be added to the `hw_constants.py` file if this answer is preferred.

### anautilities.py
There are several functions that require information about the number of VFATS and the max iEta for the gem. To start the switch to a more generalized code base, all of the functions that require those 2 pieces of info can just take the gemType as an argument with the default set to "ge11". This would not propagate the change to any part of the code, so the code in its exact form (sans the additions) would work as before.

Functions to add gemType variable in arguments (all in anautilities.py): 

* `dacAnalysis()`
* `get2DMapofDetector()`
* `DacAnalysis()`
* `get2DMapOfDetector()`
* `getEmptyPerVFATList()`
* `getMapping()`
* `getPhaseScanPlots()`
* `make2x4Canvas()`
* `make3x8Canvas()`
* `parseCalFile()`
* `parseArmDacCalFile()`
* `saveSummary()`
* `saveSummaryByiEta()`
* `getSinglePhaseScanPlot()`

Some functions in anautilities.py call other functions in the files, so these functions would have to add gemType to the argument line (not necessary, but allows for ease in future development):

* `getSinglePhaseScanPlot()`
* `saveSummaryByiEta()`
* `dacAnalysis()`
* `saveSummary()`

Note: to have gemType passed as an argument to these functions means for the strange addition of "`gemType = gemType`" in the argument line. It works, but looks very strange.

### fitScanData.py
For the file, `fitScanData.py`, many of the functions have 24 hard coded in, so as in the above cases, gemType is given to these functions and modified to get the correct value. A clean way is to just pass `nVFats` as an argument to the constructor of the `ScanDataFitter` and have it be an additional variable in the class so all 24's are replaced with `self.nVFats`. It would default to 24 for backwards compatibility

## SCurve specific changes
The next question in this issue is scope—the issue was to address the function `anaUltraScurve.py` and if we wish to only change that, we must

1. Make the following changes to scurveAlgos.py file:
  * change all instances of 24 to `vfatsPerGemVariant[gemType]`
  * change all instances of 3072 to `vfatsPerGemVariant[gemType]*128`
[... 63 lines cut ...]

With these changes, only the scurve program would be fixed. Thus comes the question of scope. This process could be expanded to fix the other programs in the gemplotting suite. These files that rely on programs needing gemType and possible more extensive changes are:

* `anaDACScans.py`
* `ana_scans.py`
* `timeHistoryAnalyser.py`
* **`anaSBitReadout.py`**
* **`anaXDAQLatency.py`**
* **`threshAlgos.py`**
* `makePhaseScanPlots.py`
* **`anaSBitMonitor.py`**
* **`plotSCurveFitResults.py`**
* **`plotSbitThreshComp.py`**
* **`latAlgos.py`**
* `fitScanData.py`
* **`summary_plots.py`**
* **`gemPlotter.py`**

If this was to be segmented, the natural break point would be files that rely on saveSummary and makeAxBCanvas, using the merged version used for anaUltraScurve.py in them. These files are the ones bolded. (Note: changes needed are not fully understood yet considering most of this was gleaned from searching these files for mentions of functions in `anautilities.py`. These will require much testing to correctly fix.)

This issue could turn into a PR for the anaUltraSCurve and naturally expand to encompass these other files, but it will require time to learn and test each of these different pieces of software. 

----

Let me know if there are problems with this plan, changes that should be made, or clarifications needed. At the very least, the code compiles and works as intended to the best of my knowledge.

bdorney, 2019-08-01:
> To aid in readability, this could be renamed to be something more descriptive like "`chamber_gem2vfatPos2PadIdx`" or something. This gets clunky very quickly, so suggestions are welcome!! Right now, [...]

Would prefer *not* changing the name of this dictionary.  Also as an aside note the naming convention that @jsturdy prefers is camelCase except for acronyms.  If an acronym is at the start of a text string it should be all lower case, e.g see how `VFAT` and `TTC` acronyms are handled:

`chamber_vfatPos2PadIdx`
`ttcGeneratorState`

But if the acronym is inside the camelCase it should be all caps, i.e. `stateOfTTCGenerator` here `TTC` is all caps.

> * `anautilities.py`: `make3x8Canvas()`
> * `anautilities.py`: `saveSummary()`

See my comments on these two below.

> Also, the max iEta and iPhi numbers are needed for setting up the canvases as well as looping correctly. This could be found by getting this from the length of `chamber_iEta2VFATPos` dictionary, but [...]

I strongly advise (read "it would not be allowed") against introducing additional hard coded parameters when it is possible to determine this from already available constants.  Consider instead:

```python
chamber_maxiEtaiPhiPair = {}
for gemType in gemVariants.keys():
   chamber_maxiEtaiPhiPair[gemType] = (len(chamber_iEta2VFATPos[gemType]), len(chamber_vfatPos2iEtaiPhi[gemType]/len(chamber_iEta2VFATPos[gemType]))
```

This would provide you with the information you want but doesn't require *any* additional hard-coded information.

> ### anautilities.py
> There are several functions that require information about the number of VFATS and the max iEta for the gem. 
> ...
> ...
> * `make2x4Canvas()`
> * `make3x8Canvas()`
> ...
> ...
> * `saveSummary()`
> * `saveSummaryByiEta()`

Again see my comments below.

> 
> Some functions in anautilities.py call other functions in the files, so these functions would have to add gemType to the argument line (not necessary, but allows for ease in future development):

If the argument is not necessary do not add it; this clutters the code.  Only add it where it *is* necessary.

> Note: to have gemType passed as an argument to these functions means for the strange addition of "`gemType = gemType`" in the argument line. It works, but looks very strange.

I guess by "argument line" you mean when the function is called.  If this is the case then I would say this is *not* strange, the LHS of that expression is a keyword reference in the calling function and the RHS is the value to be assigned.  There's nothing wrong with this.

> ### fitScanData.py
> For the file, `fitScanData.py`, many of the functions have 24 hard coded in, so as in the above cases, gemType is given to these functions and modified to get the correct value. A [...]

Passing `nVFATs` to the constructor is fine but this *also* needs to be applied to the function defined in this file:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/62b4d47bcb16498774867407bafec8783adc9992/fitting/fitScanData.py#L410-L425

Keep in mind that the class and function declared here are also used by files in the `vfatqc` repo.

> 
> ## SCurve specific changes
> The next question in this issue is scope—the issue was to address the function `anaUltraScurve.py` and if we wish to only change that, we must

While @sbutalla only mentioned `anaUltraScurve.py` the issue is really the to ensure compatibility of the *entire* package.  Partial re-factoring would be ill-advised (read "not allowed").  Prefer addressing the entire framework in one go.  This could be split across multiple developers but there should be a single PR to the central repo.

> ### makeAxBCanvas Proposal
> To expand on the changes to saveSummary and makeCanvas functions, the makeCanvas function can be generalized, but at the cost of more specification, ie the dimensions of [...]

This doesn't follow my original instructions [here](https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/226#issuecomment-505499089), specifically I was hoping for:

> > Perhaps the best solution is that saveSummary and saveSummaryByiEta absorb the functionality of makeAxBCanvas be changed to getSummaryCanvas and getSummaryCanvasByiEta take the gemType and detType arguments then when they are called they determine internally which type of plot should be corrected; following which mapping; and then return the TCanvas object appropriately and only act. There could also be a boolean argument for controlling whether or not the TCanvas is written to disk. This could then replace all calling cases.

[... 108 lines cut ...]
> On that note, there is an issue in my tests with plotting "h2ScurveSigmaDist_All.png", the candle1 draw setting gave some `Compute_Integral: Integral = zero` because the first vfat bins of this plot [...]

Naively I would assume you did not create this or the histograms for `h2ScurveMeanDist_All.png` and `h2ScurveEffPedDist_All.png` correctly...if you correctly handled the VFAT number then there should have not been empty bins, but less bins, e.g. 12 vs 24 bins for GE2/1 vs. GE1/1.  Please dobule check.

> 
> ## Scope
> With these changes, only the scurve program would be fixed. 

As discussed above a partial re-factoring would not be acceptable.  For example consider how to compare scurve results; with your changes does `plotScurveFitResults.py` work?  What about `ana_scans.py scurve [args]`? I suspect no...

[... cut ...]
> Add gemType=gemType to all anautilities.py function

Are a pretty good guiding star but be careful for locations where 3 iphi sectors were assumed, e.g.

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/62b4d47bcb16498774867407bafec8783adc9992/utils/scurveAlgos.py#L523-L525

Additionally when summary distributions are created that represent "Observable vs. vfatN` ensure that the x-axis reflects the `vfatsPerGemVariant[gemType]` and not a hangover of 24 as a max on the x-axis.

If I think of additional comments I will write them.

dteague, 2019-08-02:
Thank you for the detailed comments! I will make these changes and give an update soon as I fix/investigate things:

> Also as an aside note the naming convention that @jsturdy prefers is camelCase except for acronyms. If an acronym is at the start of a text string it should be all lower case, e.g see how VFAT and TTC acronyms are handled:

> If the argument is not necessary do not add it; this clutters the code. Only add it where it is necessary.

> Keep in mind that the class and function declared here are also used by files in the vfatqc repo. (in ref to fitScanData.py)

> Prefer addressing the entire framework in one go.

> Indeed; I've removed usage of it since I wasn't sure it would even work. This functionality could probably be removed (read "should be removed").

> This is outside the scope of your (@dteague). But there's nothing to stop you from assuming these branches exist during your development.

> For short term you can use @lmoureaux's old PR to generate a mapping text file:

> Are a pretty good guiding star but be careful for locations where 3 iphi sectors were assumed, e.g.

Noted

> I strongly advise (read "it would not be allowed") against introducing additional hard coded parameters when it is possible to determine this from already available constants. 

I've implemented that [HERE](https://gist.github.com/dteague/f36caac6aa457e8bcbf7061d41ff744b). For the `chamber_vfatPos2PadIdx`, I've implemented a little algorithm: it works, but there is probably a more efficient solution. This can be revisited.

> This doesn't follow my original instructions here, specifically I was hoping for:

Sorry for the misunderstanding, I will make the changes more to those specifications and update the gists I've posted.

> I'll be honest this PanPin functionality was implemented by a previous student who has since left the project and hasn't really been tested...additionally I never liked this split. I would not worry [...]

I'll just do some tests, make sure this suspicious code doesn't cause things to crash and minimally fix it.

> Naively I would assume you did not create this or the histograms for h2ScurveMeanDist_All.png and h2ScurveEffPedDist_All.png correctly...if you correctly handled the VFAT number then there should have not been empty bins, but less bins, e.g. 12 vs 24 bins for GE2/1 vs. GE1/1. Please dobule check.

The code does produce those plots, and it has 12 bins, but they have empty entries [image] 

It has the right format, but the first two bins are empty and gives a repeated version of the error below. Drawing this specific graph without the "candle1" option (say "colz"), it doesn't give the following error.
```sh
Error in <TH1D::ComputeIntegral>: Integral = zero
```
I will investigate this more and hopefully find the root of this problem.

----

I will make the changes Noted above and update this issue as the code changes become more mature and expand to the rest of the repo. I can also give more detailed comparisons of the old and new output for validation purposes.

commit 8c844e35 (2019-11-07): Ge21 generalizing (#252)

Closes #226 

Adaptations for generic usage of plotting tools with GE1/1 and GE2/1 electronics

* changed print to `print()` and change `%()` to `{}.format`, removed extra `print`s
* made `nVFATS` variable to pull out dictionary and made `maxChans` variable
* removed `gemType` dependence from `dbutils`
closed 2019-11-07 by jsturdy