cms-gem-daq-project/gem-plotting-tools #228: [Maintenance/Refactoring] Unify String Formatting In All Files

bdorney, 2019-06-27:
## Brief summary of issue

Spurred by discussion with @jsturdy we should unify string formatting to use `str.format` rather than `%` operator.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)
- [X] Maintanence

## Expected Behavior

All string formating and file I/O operations with formatted strings should default to using `str.format` rather than `%` operator.

## Current Behavior

Older code blocks still have statements like:

```python
"Hi I'm %i and %f and %s"%(1, 2.0, "string")
```

Instead of using `str.format`, e.g.

```python
"Hi I'm {0} and {1} and {2}".format(1, 2.0, "string"))
```

## Context (for feature requests)

Consistency.

jsturdy, 2019-06-27:
> Instead of using `str.format`, e.g.
> 
> ```python
> "Hi I'm {0} and {1} and {2}".format(1, 2.0, "string"))
> ```

This is `python2.6` compatibility oriented, which I don't feel like making an effort to support.
In `python2.7` the index is not required

In any event, the person implementing this should definitely read about how to control the formatting.

bdorney, 2019-06-27:
> This is `python2.6` compatibility oriented, which I don't feel like making an effort to support.
> In `python2.7` the index is not required
>
> In any event, the person implementing this should [...]

I'm not so sure we should abandon indexing blindly as a "this is py2.6" reason; there are certain instances in which the indexing is specifically needed to get the string sequence *correct*:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/135f4d8221333a0c886772414df50ceef2511893/utils/anautilities.py#L155-L157

(just the first example I found, here indexing is mandatory).

I would rather say that indexing can be dropped when it is not needed.

jsturdy, 2019-06-27:
> I would rather say that indexing can be dropped when it is not needed.

indeed, this was the point of my last comment.
a simple migration from the `%s` to `{}.format()` doesn't care about indexing in `>python2.7`, but more complicated formatting *can* be done, which was not possible before

lpetre-ulb, 2020-05-19:
Some (all?) of this formatting has been done in #252. The remaining statements do not deserve a fix in legacy.

closed 2020-05-19 by lpetre-ulb