cms-gem-daq-project/gem-plotting-tools #245: Add scipy to necessary modules

bregnery, 2019-08-27:
## Brief summary of issue
I would like to add scipy as a necessary module in order to use its interpolate functions with a program I am creating

### Types of issue
- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

bdorney, 2019-08-27:
Probably what would be needed here would be to provide justification for why this package is needed (or what benefit it brings), outline an example use case; provide alternatives to the `scipy` package.  Then provide a pro's and con's to each.  This would probably enable @jsturdy and @mexanick to make an informed discussion.

Recall we hashed these out in private messages on mattermost so you can use that to give justification.

bregnery, 2019-08-27:
I want to use scipy.interpolate.interp1d to interpolate a cluster charge MPV as a function of effective gain. This is a 1D interpolation.

There's is an equivalent numpy function, but it only offers 1D interpolations:
https://docs.scipy.org/doc/numpy/reference/generated/numpy.interp.html

numpy is already a dependency of gem-plotting-tools but it may not be the best approach. Brian in the past used the version in scipy previously because it gave 2D interpolation.

jsturdy, 2019-08-27:
Do you require a specific version of `scipy`? Or is the version that is installed on the DAQ machines sufficient (`0.12.1`)?

lpetre-ulb, 2020-05-18:
If I'm not mistaken, the feature (inflection point finder) has been implemented without the need of `scipy`. Feel free to re-open the issue if the dependency exists, but is not added to the requirements file/RPM packages.

closed 2020-05-18 by lpetre-ulb