cms-gem-daq-project/gem-plotting-tools #250: [bug report] Issue with multiprocessing

jsturdy, 2019-10-02:
## Brief summary of issue
It seems that between `` and `` a new behaviour was introduced, either in a return type or in how the multiprocessing is done.
Need to understand which commit brought the regression, and if it is a bug, or an expected change that needed changes elsewhere.

### Types of issue
- [x] Bug report (report an issue with the code)

## Expected Behavior
No errors

## Current Behavior
```
ana_scans.py scurve -s 2019.10.01.19.41 -c --chamberConfig --medium
Launching scurve analysis processes, this may take some time, please be patient
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 812, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 765, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib64/python2.7/multiprocessing/pool.py", line 376, in _handle_results
    task = get()
UnpicklingError: NEWOBJ class argument has NULL tp_new
```

### Steps to Reproduce (for bugs)
1. run `ana_scans.py scurve -s 2019.10.01.19.41 -c --chamberConfig --medium` on data taken at QC8

## Possible Solution (for bugs)

## Your Environment
* Version used: `0e47e29`

AndrewLevin, 2019-10-02:
`cx_Oracle` throws exceptions objects that cannot be pickled. The best solution I can think of is to catch the exception and rethrow it as an exception that can be pickled. I would propose we place a try, catch block around this line:

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/develop/utils/dbutils.py#L105

jsturdy, 2019-10-09:
OK, I think as this is non-reproduced, your solution to rethrow the exception in a way that is more robust is indeed the best we can currently do

closed 2019-10-14 by jsturdy