cms-gem-daq-project/gem-plotting-tools #271: [Bug report] OverflowError on VFat ID in anaUltraScurve.py

cgalloni, 2019-11-29:
## Brief summary of issue

While running the `anaUltraScurve.py`, we encounter problems due to the VFAT ID type which seems to give compatibility issues between the branch in the tree and the type expected from the macro `/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/scurveAlgos.py `.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

I should produce the output without any error

## Current Behavior

### Steps to Reproduce (for bugs)

On `gem904daq01`, running the commands :
```
cd /data/bigdisk/GEM-Data-Taking/GE21_Integration/GE11-X-X-XXX/scurve/2019.11.29.15.19
anaUltraScurve.py --calFile=/home/gemuser/ge11_Camp112019_calFile_calDAC.txt SCurveData.root short
```
produces the error statment:
```
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/anaUltraScurve.py", line 48, in <module>
    args.func(args,args.infilename,args.calFile,args.GEBtype,filePath,vfatList)
  File "/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/scurveAlgos.py", line 608, in anaUltraScurve
    vfatID[0] = dict_vfatID[vfat]
OverflowError: unsigned long is less than minimum
```

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

* Version used: gemuser@gem904daq01
* Shell used:

lpetre-ulb, 2019-11-29:
Are you using VFATs without Reed-Muller encoding of the chip ID? If so, it looks like the error is caused by [this issue].

mexanick, 2019-11-29:
it is exactly that. I developed an easy hack which should cover it, will open a PR after test (just a simple try-except with assigning 0 by default, as the value in case of failure is anyway unphysical)

lpetre-ulb, 2019-12-03:
The problem with that kind of easy hack is that it must be applied everywhere an overflow error could happen. Simply fixing `anaultraScurve.py` is easy, but is there any other places where the same overflow issue could occur?

lpetre-ulb, 2020-05-18:
The recommended solution is to update ROOT and PyROOT as explained in #244.

closed 2020-05-18 by lpetre-ulb