cms-gem-daq-project/gem-plotting-tools #275: SBit threshold analysis crashes when one or more vfats are masked during the scan

mexanick, 2019-12-13:
## Brief summary of issue

@bregnery 
During analysis of per-VFAT scan the code fails to find the inflection point and crashes in case one or more VFATs were masked during the scan (i.e. no data is provided)
Error message:
```bash
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/anaSBitThresh.py", line 95, in <module>
    scandate = scandate)
  File "/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/threshAlgos.py", line 1409, in sbitRateAnalysis
    dict_dacInflectPts[dacName][ohKey][vfat] = findInflectionPts(dict_Rate1DVsDACNameX[dacName][ohKey][vfat])
  File "/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/anautilities.py", line 491, in findInflectionPts
    grad = np.gradient(y, x)
  File "/home/gemuser/.local/lib/python2.7/site-packages/numpy/lib/function_base.py", line 1009, in gradient
    if (diffx == diffx[0]).all():
IndexError: index 0 is out of bounds for axis 0 with size 0
```
### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

This is a place for discussion. Definitely, the analysis should not crash, but somehow the user should be notified of a potential problem
## Current Behavior

The non-empty VFAT mask is not used during inflection point search thus leading to the crash when no data is stored for a given VFAT
### Steps to Reproduce (for bugs)

Run analysis on any per-VFAT sbitRate scan with masked VFAT(s)
## Possible Solution (for bugs)

Use the previously determined in the routine non-zero VFAT mask
## Context (for feature requests)

Impossible to determine the threshold values for a chamber in case one or more chips aren't working.
## Your Environment

* Version used:
Name        : gempython_gemplotting
Arch        : noarch
Version     : 1.5.4
Release     : 1.0e47e29git.centos7.python2.7
Size        : 1.3 M
Repo        : installed
From repo   : gemos-test
* Shell used: bash

bregnery, 2019-12-13:
This was fixed in Pull Request #268 

It has not been merged yet

jsturdy, 2019-12-13:
#268 has been merged, if this issue no longer is present, we can close

closed 2019-12-13 by mexanick