cms-gem-daq-project/gem-plotting-tools #276: Mapping change for GE21 problem

mexanick, 2019-12-13:
## Brief summary of issue

After changing chamber_iEta2VFATPos to be a list of dicts with keys `["gemType"]` not all the analysis tools are working. E.g. anaSBitReadout.py is broken now
### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior

### Steps to Reproduce (for bugs)

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

* Version used:
* Shell used:

mexanick, 2019-12-17:
The example of working patch would be 
```python
147c147
<         etaphi_to_vfat[i+1] = {row:ieta for ieta,row in chamber_iEta2VFATPos[i+1].iteritems()}
---
>         etaphi_to_vfat[i+1] = {row:ieta for ieta,row in chamber_iEta2VFATPos["ge11"][i+1].iteritems()}
300,301c300,301
<             eta = vfat_to_etaphi[vfatN[0]][0]
<             phi = vfat_to_etaphi[vfatN[0]][1]
---
>             eta = vfat_to_etaphi["ge11"][vfatN[0]][0]
>             phi = vfat_to_etaphi["ge11"][vfatN[0]][1]
398c398
<                            trimPt=None, drawOpt="", gemType=gemType, write2Disk=True)
---
>                            drawOpt="", gemType=gemType, write2Disk=True)
400c400
<                            trimPt=None, drawOpt="", gemType=gemType, write2Disk=True)
---
>                            drawOpt="", gemType=gemType, write2Disk=True)
402c402
<                            trimPt=None, drawOpt="", gemType=gemType, write2Disk=True)
---
>                            drawOpt="", gemType=gemType, write2Disk=True)
404c404
<                            trimPt=None, drawOpt="", gemType=gemType, write2Disk=True)
---
>                            drawOpt="", gemType=gemType, write2Disk=True)
```
E.g. we should either add a new parameter, or provide configuration files with all relevant info.

jsturdy, 2019-12-18:
would this change (regardless of how implemented) have an impact outside of `gemplotting`?

mexanick, 2019-12-18:
I don't think so, as this tool is not used in other pipelines