cms-gem-daq-project/gem-plotting-tools #278: chConfig produced file not compatible

giovanni-mocellin, 2020-01-16:
## Brief summary of issue
Running the analysis of scurves (in QC7 machine) produces now chConfig files that contain the column "mask" for the channels the booleans "True" and "False", instead of 1 and 0.
This makes impossible a configuration with chConfig in QC8, since confAllChambers crashes requiring a number and not a boolean.

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior
The chConfig files produced by the scurves analysis should have the "I:mask" column with numbers 0 or 1.

## Current Behavior
The chConfig files produced by the scurves analysis now have the "I:mask" column with "False" or "True".

### Steps to Reproduce (for bugs)

1. Analyze an scurve with QC7 machine
2. Check the chConfig file produced
3. Look at the column "I:mask"
4.

## Possible Solution (for bugs)
Make sure that the code is writing the boolean for the mask as an integer number

## Context (for feature requests)
When analyzing a single scurve, one can use "sed" command to change that and then anything that uses the chConfig works.
But, if one tries to run the trimming, it might get stuck since it cannot configure the chambers with the chConfig derived from the previous scurve.

## Your Environment

* Machine used: qc7daq machine
* Shell used: bash

jsturdy, 2020-01-16:
@giovanni-mocellin are the package versions (with `git` hash) the same on both machines?
@AndrewLevin, was this a recent change that modified the format of the output file?

lpetre-ulb, 2020-01-17:
I believe the issue is due to conversion from the % string formatting to the `format()` function as part of [the GE2/1 generalizing PR](https://github.com/cms-gem-daq-project/gem-plotting-tools/pull/252):

https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/bbc3e2531ac0c7c8b817768bc2f99f8dee9e9ab4/utils/scurveAlgos.py#L905-L913

Since the file `chConfig` is designed to be parsed by ROOT (as the header suggests), (all) the parameters should probably have an appropriate *format_spec* field.

jsturdy, 2020-01-17:
@dteague do you think you could address this quickly?
I guess all types should be numerical or string (though `ROOT` does support a boolean type, I'm not sure how/if `pyROOT` translates those...)
In any event, the `mask` column is currently being specified with the `I` so it's trying to interpret it as an integer.
It would be best to find all places where these `csv` `TTree` files are written and check with how they are read back, and just match the types

dteague, 2020-01-17:
Looking into this now!

commit dd086f6c (2020-01-22): Merge pull request #279 from dteague/chconfig_fix

Changed formats to match `ROOT` heading to file

Fix #278
closed 2020-01-22 by jsturdy