cms-gem-daq-project/gem-plotting-tools #290: Scurves analysis failing on QC7

cgalloni, 2020-02-13:
## Brief summary of issue

We tried to analize scurves from the QC7 or QC8 machinse both with `ana_scans.py` and directly `anaUltraScurve.py` with but the code failed. 

With `ana_scans.py` the error reported is: 
```
ana_scans.py scurve -i /data/bigdisk/GEM-Data-Taking/GE21_Integration//GE11-X-L-DS-0001/scurve/2020.02.12.18.25/SCurveData.root --chamberConfig --medium
Launching scurve analysis processes, this may take some time, please be patient
Caught <type 'exceptions.RuntimeError'>: No external mapping provided and GEB type was not recognized, terminating workers
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/ana_scans.py", line 579, in scurveMultiProcessing
    ).get(7200) # wait at most 2 hours
  File "/usr/lib64/python2.7/multiprocessing/pool.py", line 554, in get
    raise self._value
RuntimeError: No external mapping provided and GEB type was not recognized
Analysis Failed
```

With anaUltraScurve.py run from the directory:
```
anaUltraScurve.py  --calFile=/data/bigdisk/GEM-Data-Taking/GE21_Integration/GE11-X-L-DS-0001/calFile_calDac_GE11-X-L-DS-0001.txt SCurveData.root short 
Storing Output Data
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/anaUltraScurve.py", line 48, in <module>
    args.func(args,args.infilename,args.calFile,args.GEBtype,filePath,vfatList)
  File "/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/scurveAlgos.py", line 612, in anaUltraScurve
    vfatID[0] = dict_vfatID[vfat]
OverflowError: unsigned long is less than minimum
```

@mexanick used an old version of the latter from the `daq01` machine and it worked fine. 

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

Should produce the analyzed output
## Current Behavior

### Steps to Reproduce (for bugs)

1. log as gemuser on the QC& or QC8 machine
2. ana_scans.py scurve -i /data/bigdisk/GEM-Data-Taking/GE21_Integration//GE11-X-L-DS-0001/scurve/2020.02.12.18.25/SCurveData.root --chamberConfig --medium
3.
4.

## Possible Solution (for bugs)

## Context (for feature requests)

## Your Environment

`gemuser`
* Version used:
* Shell used: bash

lpetre-ulb, 2020-02-13:
> With `ana_scans.py` the error reported is:
>
> ```
> ana_scans.py scurve -i /data/bigdisk/GEM-Data-Taking/GE21_Integration//GE11-X-L-DS-0001/scurve/2020.02.12.18.25/SCurveData.root --chamberConfig [...]

Do you happen to know the GEB type? I guess it is picked up automatically somewhere, but it is neither "short" nor "long".

> With anaUltraScurve.py run from the directory:
>
> ```
> anaUltraScurve.py  --calFile=/data/bigdisk/GEM-Data-Taking/GE21_Integration/GE11-X-L-DS-0001/calFile_calDac_GE11-X-L-DS-0001.txt SCurveData.root [...]

Do you use VFAT3 hybrid v2 without the Reed-Muller encoding? If yes, an updated version of PyROOT (as so ROOT) is required; see https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/244.

cgalloni, 2020-02-14:
GEB type used is "short". 

> Also, does the `DATA_PATH` points to `/data/bigdisk/GEM-Data-Taking/GE21_Integration/`; I may have an idea if it doesn't.

Yes this was done on purspose at this point

lpetre-ulb, 2020-02-14:
How was the issue fixed? Are you just working around it?

> GEB type used is "short".

If it is `short` is really the value of `GEBtype` at this point in the code, the issue looks genuine:
https://github.com/cms-gem-daq-project/gem-plotting-tools/blob/e86ae2a09c843b917540f4a6648860ae47d0b20f/utils/scurveAlgos.py#L298-L308

cgalloni, 2020-02-14:
No, sorry I just pressed the wrong button. 
Yes we used the VFAT3 hybrid v2 without the Reed-Muller encoding, so we configured them with a slightly modified version of `getCalInfoFromDB.py` where we feed the chipId by hand.

We used the  `DATA_PATH`  as `/data/bigdisk/GEM-Data-Taking/GE21_Integration/`   not to interefere with other operations of the QC8.
Then we proceed as:
```
python getCalInfoFromDB.py 2 5 0 --write2File --write2CTP7
testConnectivity.py 2 5  0x1 --gemType ge11 --detType short -a -i --acceptBadDACFits --skipScurve
```
Because the scurve part wouldn't work, we decided to run it then on its own. We tried then both the commands above. 
Maybe the fact that the chamber name contains the "L" and  we used short in the configuration is 
creating the problem. We are taking a latency  scan, then let me try to rename the chamber and  try taking scurves again.

cgalloni, 2020-02-14:
I changed the chamber name in `editConfig.py` to be consistent with being a short chamber and took scurves again, but still the analysis part fails :
```
cd /data/bigdisk/GEM-Data-Taking/GE11_QC8/GE11-X-S-XXX-0001/scurve/2020.02.14.16.23

anaUltraScurve.py  --calFile=/data/bigdisk/GEM-Data-Taking/GE11_QC8//GE11-X-S-XXX-0001/calFile_calDac_GE11-X-S-XXX-0001.txt   SCurveData.root short
```
with the error: 
```
Removing Hot Channels from Output Histograms
Storing Output Data
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/anaUltraScurve.py", line 48, in <module>
    args.func(args,args.infilename,args.calFile,args.GEBtype,filePath,vfatList)
  File "/usr/lib/python2.7/site-packages/gempython/gemplotting/utils/scurveAlgos.py", line 612, in anaUltraScurve
    vfatID[0] = dict_vfatID[vfat]
OverflowError: unsigned long is less than minimum
```

lpetre-ulb, 2020-02-14:
Okay, that error is already reported and described in details in issue https://github.com/cms-gem-daq-project/gem-plotting-tools/issues/244. The ROOT version installed on the QC7 DAQ machine is 6.16.00 while the fix is implemented in 6.18.02. The best way to fix your issue would be to update the software.

lpetre-ulb, 2020-02-27:
@cgalloni, any update on this issue? Is it relevant? Does the newer version of ROOT fix it? I don't feel confident changing how the chamber name and/or chamber configuration file are parsed.

cgalloni, 2020-02-28:
I think un update of ROOT on the machine is needed to test out if it solves the issue. 
Haven't tried it yet.

lpetre-ulb, 2020-05-19:
Probably a duplicate of #244 for which a ROOT update is required.

closed 2020-05-19 by lpetre-ulb