cms-gem-daq-project/gem-plotting-tools #293: cx_Oracle not available in repositories

lpetre-ulb, 2020-03-23:
## Brief summary of issue

Sven attempted to install the `gempython_gemplotting` package from the `gemos` repositories on a pristine computer. That resulted in an error about the `cx_Oracle` package not being found.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

The legacy GEM software from the repositories should install seamlessly on any CC7 machine.

## Current Behavior

The `cx_Oracle` package is missing; it cannot be found in neither the GEM repositories nor in the standard repositories under the name `cx_Oracle`.

### Steps to Reproduce (for bugs)

1. Follow the instructions from [the TWiki page](https://twiki.cern.ch/twiki/bin/view/CMS/GEMOnlineSoftwareDevelopment)
2. `sudo curl -L https://cern.ch/test-gemdaq-ci/sw/gemos/repos/releases/legacy/gemos_release_centos7_x86_64.repo -o /etc/yum.repos.d/gemos.repo`
3. `sudo yum install -y xhal reedmuller wiscrpcsvc gempython\*`

## Possible Solution (for bugs)

Provide a pre-compiled `cx_Oracle` package in the `gemos-extras` repository.

## Your Environment

* Version used:
* Shell used:

jsturdy, 2020-03-24:
The twiki page was not correct but has since been updated to add the instruction that was listed in the email thread, i.e., `--enablerepo=cernonly`

lpetre-ulb, 2020-03-24:
I guess this repository is only available in CC7? While I have good a experience with CentOS, it is more mitigated with CERN CentOS... If CC is needed at CERN, it isn't for remote test stands. Would it possible to target vanilla CentOS and provide the required packages?